Lora-net/lora_gateway #4: Timing synch accuracy and dev board access?

rbm7, 2015-04-21:
HI Matthieu, I stumbled across this project and I wonder if you could tell me more about it? Specifically, I'm wondering when the dev board for the 1301 that you're using (or are you primarily targeting the kerlink module?) will be available to users outside Semtech? Also, I see that you've added GPS synchronization to the code; what is the absolute time-resolution of the received time-stamps (the time-stamp of a received message from a remote LoRa wireless node)?

Cool project! Thanks - Rob

Limro, 2016-05-10:
Very good question - I would like to know that too.

According to [the readme](https://github.com/Lora-net/lora_gateway/tree/master/libloragw), section 4.3:

> the host must be connected to a GPS receiver via a serial link

Serial links are rather slow (milliseconds), as far as I know.

alex-fltm, 2016-05-10:
The 1301 board can support the PPS signal from GPS in order to increase the accuracy of the timestamps. 

> In 'triggered' mode (aka PPS/GPS mode), the packet, typically a beacon, is emitted 1.5ms after a rising edge of the trigger signal. Because there is no way to anticipate the triggering event and start the analog circuitry beforehand, that delay must be taken into account in the protocol.

JiapengLi, 2017-07-12:
Very interesting topic.

From my view, PPS signal can't increase the accuracy of the timestamps, the PPS signal is just to sync between GPS time and LoRa time (SX1301 time), the SX1301 time accuracy is decided 32MHz TCXO which is used to driver SX125x (should be, not very sure). And TIMESTAMP register unit is microsecond.

The PPS signal also could be used to trigger Class B beacon, this TX timing is decide be PSS accuracy.

From LoRaWAN specification, TX packet accuracy should satisfy +/-20us. So SX1301 TX timing accuracy should be better than that. For RX timing should be also around that.

mcoracin, 2017-07-12:
GPS is only necessary for Class-B, in order to:
- trigger a beacon with PPS accuracy
- convert SX1301 to GPS time, and vice-versa, for class-B downlink.

The SX1301 timestamp has a resolution of 1µs, as said by @JiapengLi 

I think that this topic is old and has been answered, so I close it. Don't hesitate to re-open if more question.

closed 2017-07-12 by mcoracin

blinkomaniak, 2019-09-02:
Hi everyone

I want to know if it is possible that the SX1301 obtains nanosecond accuracy from the GPS by modifying the **[timersync.c](https://github.com/Lora-net/packet_forwarder/blob/master/lora_pkt_fwd/src/timersync.c)** file. Also, does any body knows in what consists the license that Semtech talks about to decrypt the fine timestamps? 

[image]

I just want to know if it is possible to get some nanosecond accuracy for the timestamps using  the SX1301 and Ublox Max7Q GPS (RAK7243)

iweeks, 2019-09-04:
It is not possible to generate nanosecond timestamps by simply modifying the timersync.c file.  In order to increase timing accuracy of packet arrival, Semtech developed a DSP algorithm that runs on a dedicated FPGA in their high performance gateway reference design.

blinkomaniak, 2019-09-04:
Thanks for your help @iweeks, 
So one more question please, how can I know which LoRa Gateway comes with the new high performance reference design?. From what I've researched, they offer GPS aditional hardware but they do not specify if they support GPS nanosecond synchronization capabilities to support  TDoA-based geolocation applications. 

In order to do some proof of concept of TDoA, I bought 03 RAKs 7243 having in mind that they had GPS. I even search over the internet to modify the firmware source code to obtain the received packet metadata time in nanoseconds, but it is impossible to synchronize the 03 RAKs. Of course, this might be because they acquire their GPS time in different microseconds or even milliseconds. So right now, it seems that I am down $800 on my project because those RAKs are not useful for my TDoA proof of concept and I don't want to waste more money on hardware without having the confirmation that they are Semtech v2 gateways. As a matter of fact, I don't even know with total certainty that those RAKs SX1301 are not Semtech v2. How can I know that? 

Thanks

iweeks, 2019-09-05:
Commercially available V2.1 geolocation gateways will contain an additional FPGA called SX1388 which is configured with the nano second packet detection algorithm developed by Semtech.  Existing GW manufacturers with licenses for this functionality include Multitech, Kerlink & Cisco... I'm sure there are also others available in the market by now.