Lora-net/lora_gateway #7: north american config

VirTERM, 2015-10-24:
Where do I find global_con.json suitable for North American bands?

nestorayuso, 2015-10-24:
here: https://github.com/Lora-net/packet_forwarder/tree/v1.3.0/basic_pkt_fwd/cfg
to cover all the 915 band in North America you need a gateway with an array of 8 sx1301 decoders

VirTERM, 2015-10-25:
Sorry, but I am confused as I was referring to "lora_gateway/util_pkt_logger/global_conf.json" with 2 sx1257 radios... Is there North American version of is config file?

mcoracin, 2016-04-13:
Hi VirTERM,
Is this issue still open for you or can I close it?

The lora_gateway and packet_forwarder repos target EU gateway and an example for EU 868MHz frequency plan is given.
You can adapt this configuration file to NA channels if you have the corresponding frontend.

However, as said by nestorayuso, this gateway only supports 8 channels, you won't be able to support to 64 channels for NA.

VirTERM, 2016-04-13:
all under control now…for hybrid mode (8 channels)

mcoracin, 2016-04-13:
ok thanks, so I close the issue.
sorry for not giving any answer before, I just haven't been notified about the requests up to now. :)

closed 2016-04-13 by mcoracin