Lora-net/lora_gateway #8: Fatal error: can't create obj/util_spectral_scan.o: No such file or directory

vredzhepov, 2015-10-28:
The latest version of the lora_gateway build failed. The reason of this error is there's no obj directory in the util_spectral_scan directory.

make[1]: Entering directory `/tmp/lora_gateway-master/util_spectral_scan'
gcc -c -O2 -Wall -Wextra -std=c99 -I inc -D DEBUG_SPI=0 -D DEBUG_REG=0 src/util_spectral_scan.c -o obj/util_spectral_scan.o
Assembler messages:
Fatal error: can't create obj/util_spectral_scan.o: No such file or directory
make[1]: *** [obj/util_spectral_scan.o] Error 1
make[1]: Leaving directory `/tmp/lora_gateway-master/util_spectral_scan'
make: *** [all] Error 2

JiapengLi, 2015-10-30:
Workaround:

     mkdir util_spectral_scan/obj

mcoracin, 2016-04-13:
this will be fixed with next release, thanks for reporting.

closed 2016-04-13 by mcoracin