Lora-net/lora_gateway #10: Failed in test_loragw_reg test, anyone can help?

rossihwang, 2015-12-02:
My hardware are ic880a and raspberry pi,  They are connected through SPI. I modify the lastest code a little(add the reset pin control). I tested the SPI write function with the logic analysor, it works well. But it can't pass the test_loragw_reg test, in which some read back values are fault.   

```
Beginning of test for loragw_reg.c
Note: SPI port opened and configured ok
Note: Reset pin toggle ok
Note: SPI read success
INFO: no FPGA detected
Note: SPI read success
Note: SPI write success
Note: success connecting the concentrator
Note: SPI write success
Start of register verification
Note: SPI read success
###MISMATCH### reg number 0 read: 2 (2) default: 0 (0)
Note: SPI read success
+++MATCH+++ reg number 1 read: 0 (0) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 2 read: 187 (bb) default: 103 (67)
lgw_spi_rb:383: BURST READ: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst read success
###MISMATCH### reg number 3 read: 45404 (b15c) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 4 read: 32 (20) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 5 read: 163 (a3) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 6 read: 126 (7e) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 7 read: 169 (a9) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 8 read: 176 (b0) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 9 read: 164 (a4) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 10 read: 115 (73) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 11 read: 83 (53) default: 0 (0)
lgw_spi_rb:383: BURST READ: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst read success
###MISMATCH### reg number 12 read: 6084 (17c4) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 13 read: 127 (7f) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 14 read: 66 (42) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 15 read: 1 (1) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 16 read: 1 (1) default: 0 (0)
Note: SPI read success
+++MATCH+++ reg number 17 read: 0 (0) default: 0 (0)
Note: SPI read success
+++MATCH+++ reg number 18 read: 0 (0) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 19 read: 0 (0) default: 1 (1)
Note: SPI read success
###MISMATCH### reg number 20 read: 0 (0) default: 1 (1)
Note: SPI read success
###MISMATCH### reg number 21 read: 1 (1) default: 0 (0)
Note: SPI read success
+++MATCH+++ reg number 22 read: 0 (0) default: 0 (0)
Note: SPI read success
+++MATCH+++ reg number 23 read: 0 (0) default: 0 (0)
Note: SPI read success
+++MATCH+++ reg number 24 read: 0 (0) default: 0 (0)
Note: SPI read success
+++MATCH+++ reg number 25 read: 0 (0) default: 0 (0)
Note: SPI read success
###MISMATCH### reg number 26 read: 1 (1) default: 0 (0)
Note: SPI read success
[... 636 lines cut ...]
lgw_spi_rb:383: BURST READ: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst read success
PREAMBLE_SYMB1_NB = 33272 (should be 49253)
lgw_spi_wb:321: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:383: BURST READ: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst read success
ADJUST_MODEM_START_OFFSET_SF12_RDX4 = 4410 (should be 3173)
lgw_spi_wb:321: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:383: BURST READ: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst read success
IF_FREQ_1 = -1782 (should be -1947)
lgw_spi_wb:321: BURST WRITE: to trans 256 # chunk 256 # transferred 256 
Note: SPI burst write success
lgw_spi_rb:383: BURST READ: to trans 256 # chunk 256 # transferred 256 
Note: SPI burst read success
Note: SPI port closed
Note: success disconnecting the concentrator
End of test for loragw_reg.c
```

Aslo I can ensure that the interface is connected well, 

thanks!

bitmorse, 2015-12-05:
I'm trying to get the same test to work and have the same hardware. I'm stuck with the ic880a not responding. Could you please share how you got the reset to work?

rossihwang, 2015-12-07:
hi @bitmorse  . I use the wiringPi library to do so, though it is not a pretty way, because you need root to run the application.

first, you should install wiringPi.
then add some code at the end of the lgw_spi_open() in loragw_spi.native.c

```
#include <wiringPi.h>

int lgw_spi_open() {
    
    ...

    wiringPiSetup();
    pinMode(31, OUTPUT); // You can use other pin
    digitalWrite(31, HIGH);
    // Maybe need some delay here. Delay need to be more than 100ns.
    digitalWrite(31, LOW);
    DEBUG_MSG("Note: Reset pin toggle ok\n");

    return LGW_SPI_SUCCESS;
}

```

remember to add -lwiringPi to you LIBS in Makefile.

That is it!

bitmorse, 2015-12-07:
Thank you very much! Which hardware reset pin did you connect to on the
iC880a though? There seem to be 2.

bitmorse, 2015-12-08:
@rossihwang after setting k.cs_change=0 everywhere and then trying the pkt_logger, everything works fine. we can also verify that setting the SPI reset pin first is necessary for things to work.

rossihwang, 2015-12-08:
@bitmorse Really?! I would have a try!
Actually I do the reset because the IMST engineerer told me that.
BTW, Do you modify the PCB according to the file "iC880A_AN014_USB_modifications_for_SPI_V1_2.pdf"??

rossihwang, 2015-12-08:
@bitmorse I tried it, the "test_loragw_reg" works better but still some error!

```
Beginning of test for loragw_reg.c
Start of register verification
+++MATCH+++ reg number 0 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 1 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 2 read: 103 (67) default: 103 (67)
+++MATCH+++ reg number 3 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 4 read: 146 (92) default: 0 (0)
+++MATCH+++ reg number 5 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 6 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 7 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 8 read: 125 (7d) default: 0 (0)
+++MATCH+++ reg number 9 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 10 read: 8 (8) default: 0 (0)
+++MATCH+++ reg number 11 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 12 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 13 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 14 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 15 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 16 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 17 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 18 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 19 read: 1 (1) default: 1 (1)
+++MATCH+++ reg number 20 read: 1 (1) default: 1 (1)
+++MATCH+++ reg number 21 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 22 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 23 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 24 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 25 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 26 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 27 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 28 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 29 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 30 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 31 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 32 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 33 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 34 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 35 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 36 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 37 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 38 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 39 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 40 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 41 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 42 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 43 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 44 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 45 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 46 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 47 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 48 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 49 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 50 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 51 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 52 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 53 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 54 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 55 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 56 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 57 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 58 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 59 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 60 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 61 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 62 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 63 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 64 read: 0 (0) default: 0 (0)
[... 246 lines cut ...]
+++MATCH+++ reg number 311 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 312 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 313 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 314 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 315 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 316 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 317 read: 0 (0) default: 1 (1)
+++MATCH+++ reg number 318 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 319 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 320 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 321 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 322 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 323 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 324 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 325 read: 0 (0) default: 0 (0)
End of register verification
IMPLICIT_PAYLOAD_LENGHT = 197 (should be 197)
FRAME_SYNCH_PEAK2_POS = 11 (should be 11)
PREAMBLE_SYMB1_NB = 0 (should be 49253)
ADJUST_MODEM_START_OFFSET_SF12_RDX4 = 0 (should be 3173)
IF_FREQ_1 = 0 (should be -1947)
End of test for loragw_reg.c
```

Would you tell me how do you connect the RPi with the ic880A?

bitmorse, 2015-12-08:
did you make sure the board resets prior to spi comms? and did you set cs_change to 0 in occurrence? are you using spidev0.1 on raspi? are you using the latest lora-gateway version?

we only connected spi lines and ground. we do not use usb to power the ic880a. we reset "manually" at the moment.

rossihwang, 2015-12-09:
Think you so much. I found I didn't modify all the cs_change to 0. It works now!

abouillot, 2016-03-18:
I'm ending up in a similar situation. I've cloned the lora_gateway repositiry, changed the 4 occurrences of cs_change to 0:

```
pi@raspberrrypi:~/lora_gateway/libloragw $ grep -r cs_ *
src/loragw_spi.native.c:        k.cs_change = 0;
src/loragw_spi.native.c:        k.cs_change = 0;
src/loragw_spi.native.c:        k[0].cs_change = 0;
src/loragw_spi.native.c:        k[1].cs_change = 0;
src/loragw_spi.native.c:        k[0].cs_change = 0;
src/loragw_spi.native.c:        k[1].cs_change = 0;
```
I'm using spidev0.0, the board is reset using GPIO0 pin 11 and the outcome is similar to last @rossihwang output:

```
pi@raspberrrypi:~/lora_gateway/libloragw $ sudo ./test_loragw_reg
Beginning of test for loragw_reg.c
Start of register verification
+++MATCH+++ reg number 0 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 1 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 2 read: 103 (67) default: 103 (67)
+++MATCH+++ reg number 3 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 4 read: 217 (d9) default: 0 (0)
+++MATCH+++ reg number 5 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 6 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 7 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 8 read: 244 (f4) default: 0 (0)
+++MATCH+++ reg number 9 read: 0 (0) default: 0 (0)
###MISMATCH### reg number 10 read: 138 (8a) default: 0 (0)
+++MATCH+++ reg number 11 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 12 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 13 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 14 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 15 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 16 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 17 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 18 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 19 read: 1 (1) default: 1 (1)
+++MATCH+++ reg number 20 read: 1 (1) default: 1 (1)
+++MATCH+++ reg number 21 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 22 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 23 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 24 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 25 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 26 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 27 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 28 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 29 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 30 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 31 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 32 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 33 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 34 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 35 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 36 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 37 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 38 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 39 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 40 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 41 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 42 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 43 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 44 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 45 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 46 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 47 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 48 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 49 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 50 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 51 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 52 read: 0 (0) default: 0 (0)
[... 257 lines cut ...]
+++MATCH+++ reg number 310 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 311 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 312 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 313 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 314 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 315 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 316 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 317 read: 1 (1) default: 1 (1)
+++MATCH+++ reg number 318 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 319 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 320 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 321 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 322 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 323 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 324 read: 0 (0) default: 0 (0)
+++MATCH+++ reg number 325 read: 0 (0) default: 0 (0)
End of register verification
IMPLICIT_PAYLOAD_LENGHT = 197 (should be 197)
FRAME_SYNCH_PEAK2_POS = 11 (should be 11)
PREAMBLE_SYMB1_NB = 49253 (should be 49253)
ADJUST_MODEM_START_OFFSET_SF12_RDX4 = 3173 (should be 3173)
IF_FREQ_1 = -1947 (should be -1947)
End of test for loragw_reg.c
```
I guess I forgot something, but ran out of idea.

JiapengLi, 2016-03-19:
@abouillot which spi module did you use? spi-bcm2835 or spi-2708? Could check with **lsmod**.

abouillot, 2016-03-19:
The module is bcm2835:
```
pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
cfg80211              499834  0
rfkill                 22491  1 cfg80211
spi_bcm2835             7948  0
bcm2835_gpiomem         3703  0
snd_bcm2835            22317  3
snd_pcm                92581  1 snd_bcm2835
snd_timer              23454  1 snd_pcm
snd                    68161  9 snd_bcm2835,snd_timer,snd_pcm
uio_pdrv_genirq         3690  0
uio                    10002  1 uio_pdrv_genirq
i2c_dev                 6730  0
fuse                   91981  3
ipv6                  360374  24
```

JiapengLi, 2016-03-31:
@abouillot From my experience lora_gateway works correctly with RPi driver bcm-2708 (older than bcm-2835), no patch needed. So i guess there is different cs_change implementation between bcm-2708 and bcm-2835.

I referred this article https://www.kernel.org/doc/htmldocs/device-drivers/API-struct-spi-transfer.html, although not very easy to understand. But It seems the cs_change signal is generally used to inactive chip select signal for standard implementation. 

Not very sure, but I think your patch make sense. Thank your for the sharing!

rossihwang, 2016-03-31:
@abouillot  I also work on RPI B+, the spi is snd_bcm2835. The SPI test may not pass completly. I think you should try the ohter tests.

sylvain-miermont, 2016-03-31:
@abouillot Hi
It's ok for some registers to display a mismatch because some special registers have a value that is not always constant during initialization, but the test_loragw_reg program doesn't know if a register is special or not, and test them all sequentially.

Simply said, as soon as you get matching values on the latest section:
```
IMPLICIT_PAYLOAD_LENGHT = 197 (should be 197)
FRAME_SYNCH_PEAK2_POS = 11 (should be 11)
PREAMBLE_SYMB1_NB = 49253 (should be 49253)
ADJUST_MODEM_START_OFFSET_SF12_RDX4 = 3173 (should be 3173)
IF_FREQ_1 = -1947 (should be -1947)
```
you can assume that the SX1301 is working as intended, and SPI communication is ok. The register-by-register test that is done initially was designed for internal development and debug, and neither the documentation, nor the program itself explains how the results should be interpreted. My apologies for that.

If you have any doubt about the SPI communication reliability, or start seeing a SX1301 act *funny*, please run the util_spi_stress for a couple of minutes or hours. Since Broadcom fixed their formerly barely functional drivers, if your wiring is done properly (ground wire close to SCK/MOSI/MISO wires, length kept reasonable) your SPI link should be rock-solid.

S.

abouillot, 2016-03-31:
@sylvain-miermont Hi,

Thanks for the clarification. I have the GW functional for some days now, but was puzzled by this discrepancy in the test results. I'm reassured now assuming all is OK and will look elsewhere in case I face some issue.

mcoracin, 2016-04-13:
Hi all,

concerning possible MISMATCH returned by the test_loragw_reg binary, it is expected for some registers.
All the registers *_DATA correspond to some randomly initilialized memory, so may differ from one run to the other.

So if the MISMATCH you get are on those registers, don't worry, it is expected.

I close this issue.

JiapengLi, 2016-04-13:
@mcoracin Hi Michael, Would you please check the usage of cs_change? I get several customers report that lora_gateway cs_change usage doesn't follow Linux standard.

mcoracin, 2016-04-13:
@JiapengLi  Hi,
It is true that there is something unclear here.
The lora_gateway was initially tested with various hosts, Raspberry Pi (spi_bcm2708), PCduino, Kerlink hosts (don't know what they have inside), and it is working with all of them the way is is released.
More recently, we have seen several hosts behaving differently with the cs_change, like the RPi with spi_bcm2835 and few others, and for which reverting the cs_change to 0 fixed issues.

I could not spend time on this so far to analyze deeper the issue, and how it could be fixed for all platforms.

I'll try to check this when possible.

In the meantime, if anyone as a final answer on this, I'm open! :)

mcoracin, 2016-04-25:
Hello,

For your information, the latest release v3.2.1 comes with the cs_change set to 0.
This lets the SPI driver handle chip select.

It has been tested on Raspberry Pi with both spi_bcm2708 and spi_bcm2835 drivers, and it works with both.

Let me know if you face any issue with other SPI drivers.

Regards

closed 2016-04-25 by mcoracin