Lora-net/lora_gateway #11: Internals of "delayed" packet forwarding

ISGA, 2015-12-19:
Hi,
Could anybody point to the internals of "delayed" packet forwarding ?

I could see the timestamp handling at Packet Forwarder and passed as metadata to lora_gateway through lgw_send().
The timestamp is put at buff  (libloragw/src/loragw_hal.c)
1481         if (pkt_data.tx_mode == TIMESTAMPED)
1482         {
1483                 count_trig = pkt_data.count_us - TX_START_DELAY;
1484                 buff[3] = 0xFF & (count_trig >> 24);
1485                 buff[4] = 0xFF & (count_trig >> 16);
1486                 buff[5] = 0xFF & (count_trig >> 8);
1487                 buff[6] = 0xFF &  count_trig;
1488         }
Appreciate if any body could tell me how writing to LGW_TX_TRIG_DELAYED works :+1: 
lgw_reg_w(LGW_TX_TRIG_DELAYED, 1);
Is the send maintaining multiple buffers in the queue based on timestamp to send ? 

Thanks

jlesech, 2015-12-19:
Hi,
Timestamped TX mode works, but you can't send multiple buffers at once. If a buffer is waiting, the next will overwrite it...
It is necessary to schedule the sending of the buffer before forwarding it to the lgw_send function.

sylvain-miermont, 2016-03-31:
Hi,
"Delayed" packet sending is using hardware assistance (a 1 MHz 32-bit always-running timer) to be able to send a packet  with a very accurate delay after receiving one, because the same time is used for timestamping received packets.

The SX1301 has a single "slot" for sending packets (delayed or not), no FIFO, and the outgoing "delayed" packet must be **fully loaded** (parameters and payload) in the SX1301, **before** the internal SX1301 counter reaches the value `count_us - TX_START_DELAY`

When the counter reaches `count_us - TX_START_DELAY` exactly, the SX1301 automatically switches on/configures/calibrates/etc all the internal blocs and peripherals that are necessary to send a packet, and then, when the counter reaches `count_us` the fist sample of the first chirp of the preamble is sent by the TX modem.

S.

mcoracin, 2016-04-13:
Hi,

Concerning the TX buffering, there is an initial implementation done on the "experimental' branch of the packet_forwarder repo, You can refer to it if needed.
This branch should be ultimately merged to the 'master' branch along with Class-B support. The work is on-going.

mcoracin, 2017-07-12:
Just-in-time TX scheduling and TX buffering has been added to the packet forwarder since a while now, so I close this issue.

closed 2017-07-12 by mcoracin