Lora-net/lora_gateway #13: Version of calibration firmware not expected, actual:8 expected:2

norbertherbert, 2016-03-12:
I am trying to set up a Lora Gateway based on an *iC880A-SPI* board and a *Raspberry Pi* following the instructions of the *iC880A-SPI_QuickStartGuide.pdf* document. I have set up everything and successfully compiled the latest code (v3.2.0)
However I was not able to pass the _**test_loragw_reg**_ test just like it was written in  *https://github.com/Lora-net/lora_gateway/issues/10*. After I changed the *k.cs_change* parameter to *0* like it was written there, the test seems to run successfully.

However I still cannot start the concentrator. The error I receive is the following:
```
lgw_spi_rb:371: BURST READ: to trans 8192 # chunk 1024 # transferred 1024
lgw_spi_rb:371: BURST READ: to trans 7168 # chunk 1024 # transferred 2048
lgw_spi_rb:371: BURST READ: to trans 6144 # chunk 1024 # transferred 3072
lgw_spi_rb:371: BURST READ: to trans 5120 # chunk 1024 # transferred 4096
lgw_spi_rb:371: BURST READ: to trans 4096 # chunk 1024 # transferred 5120
lgw_spi_rb:371: BURST READ: to trans 3072 # chunk 1024 # transferred 6144
lgw_spi_rb:371: BURST READ: to trans 2048 # chunk 1024 # transferred 7168
lgw_spi_rb:371: BURST READ: to trans 1024 # chunk 1024 # transferred 8192
Note: SPI burst read success
ERROR: Failed to load fw 1
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
ERROR: Version of calibration firmware not expected, actual:8 expected:2
ERROR: failed to start the concentrator
```
Can anyone advise how I can proceed from here?

Thanks:
Norbert

rossihwang, 2016-03-12:
it may some error occurred during loading the firmware(for the mcu in the sx1301). 
Did you reset the sx1301?

norbertherbert, 2016-03-12:
Yes, I did reset it.
In the meantime I found another error message in the output:
*setup_sx125x:377: ERROR: UNEXPECTED VALUE 0 FOR RADIO TYPE*
(it was several lines before the last one)
Please find the full output below:
```
lora@raspberrypi:~ $ /home/lora/lora_gateway/util_tx_test/util_tx_test -r 1257 -f 866.5
Sending -1 packets on 866500000 Hz (BW 125 kHz, SF 10, CR 1, 16 bytes payload, 8 symbols preamble) at 14 dBm, with 1000 ms between each
lgw_board_setconf:566: Note: board configuration; lorawan_public:1, clksrc:1
lgw_rxrf_setconf:598: Note: rf_chain 0 configuration; en:1 freq:866500000 rssi_offset:0.000000 radio_type:2 tx_enable:1
Note: SPI port opened and configured ok
Note: SPI read success
INFO: no FPGA detected
Note: SPI read success
Note: SPI write success
Note: success connecting the concentrator
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
setup_sx125x:358: Note: SX125x #0 version register returned 0x21
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
setup_sx125x:366: Note: SX125x #0 clock output disabled
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
[... 121 lines cut ...]
lgw_spi_wb:309: BURST WRITE: to trans 1024 # chunk 1024 # transferred 8192
Note: SPI burst write success
Note: SPI read success
lgw_spi_rb:371: BURST READ: to trans 8192 # chunk 1024 # transferred 1024
lgw_spi_rb:371: BURST READ: to trans 7168 # chunk 1024 # transferred 2048
lgw_spi_rb:371: BURST READ: to trans 6144 # chunk 1024 # transferred 3072
lgw_spi_rb:371: BURST READ: to trans 5120 # chunk 1024 # transferred 4096
lgw_spi_rb:371: BURST READ: to trans 4096 # chunk 1024 # transferred 5120
lgw_spi_rb:371: BURST READ: to trans 3072 # chunk 1024 # transferred 6144
lgw_spi_rb:371: BURST READ: to trans 2048 # chunk 1024 # transferred 7168
lgw_spi_rb:371: BURST READ: to trans 1024 # chunk 1024 # transferred 8192
Note: SPI burst read success
ERROR: Failed to load fw 1
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
ERROR: Version of calibration firmware not expected, actual:8 expected:2
ERROR: failed to start the concentrator
lora@raspberrypi:~ $
```

norbertherbert, 2016-03-14:
In the meantime I found the solution: 
It was not enough to change the *k.cs_change* parameter from *1* to *0*
```
k.cs_change=0
```
but I had to look for *k[1].cs_change* parameter too.
```
k[1].cs_change=0
```

rossihwang, 2016-03-14:
The sx1301 wasn't selected : (

closed 2016-03-16 by norbertherbert

VanXungUIT, 2016-12-16:
Hi! 
you solved the problem yet?
I'm having problems.
Can you help me!

iBrick, 2016-12-16:
Usually this error shows that you have wrong pin connections. Check your wiring.

kaizen8501, 2018-02-14:
Hi!
I had also same problem. But I sorted this problem out after reset sx1301.
So I think you have better to check your Raspberry Pi reset pin what it is work well.