Lora-net/lora_gateway #15: SX1301 register description

x893, 2016-03-16:
Hello,
can anybody help with SX1301 register description ?

Thanks in advance

mcoracin, 2016-04-13:
Hello,

What do you need to know?
Semtech does not provide the full description of the registers, and the present code is there to show how to use it.

But if you have specific questions, I can try to answer.

Regards

x893, 2016-04-13:
Hello,
i've simple question - how to use interrupt mode for SX1301 ?

mcoracin, 2016-04-14:
Hello,
There is no interrupt mode on SX1301, you have to poll registers.

x893, 2016-04-14:
Hi,
it very strange - DIO lines exists.
Semtech looks as most china company w/o any documentation.

mcoracin, 2016-04-20:
Hello,
No interrupt mode on SX1301.
Concerning the documentation, SX1301 is an exception in the sense that it is purposely not documented.
The present source code is given as a reference of usage and configuration.

If you have specific questions about the driver code, you can ask, we'll try to answer.

Regards

closed 2016-04-20 by mcoracin

x893, 2016-04-20:
I port gateway from linux to FreeRTOS with low power MCU (STM32L and may be to EFM32) but to reduce consumption think about interrupt mode. Also i use GSM modem to provide link to internet or microSD for standalone application. Thanks for your answer about interrupt.

fanjm95, 2018-08-30:
SX1301 How to set register  Low Data Rate Optimization?

Kodest, 2019-07-01:
https://github.com/Lora-net/lora_gateway/wiki/SX1301-GPIO-output-configuration

If you wire `RX_BUFFER_NOT_EMPTY` GPIO of SX1301 to one of your MCU's GPIOs, you can spare your SPI busy loop polling and just read received LoRa data packets on a rising edge. It works, I've just tried. This also means a better response time because we are possibly not in sleep when a packet arrives but can react instantly.

ttangzr, 2020-12-02:
Hi,
I try to get the GPIO output of SX1308 pico gateway with STM32F4 MCU. I set the `LGW_GPIO_SELECT_OUTPUT` as follows:

``` C++
lgw_reg_w(LGW_GPIO_MODE, 31);
lgw_reg_w(LGW_GPIO_SELECT_OUTPUT, 4);
```

GPIOs table will be then as follow (refer to Table 7 of SX1301 DS)
- GPIO[0] -> sensor_0_detect
- GPIO[1] -> sensor_0_sync
- GPIO[2] -> sensor_0_header_val
- GPIO[3] -> 1’b0
- GPIO[4] -> 1’b0

But I have some questions:
1. Does anybody know the TYPE of these output values? Analog or digital high/low level signals?
2. What does the 1’b0 mean?

Thanks.