Lora-net/lora_gateway #16: Is hybrid mode included in this version?

sheenhx, 2016-05-30:
Hi, recently your FAE Michael Fink told us that the hybrid mode is in this github. But we can't find it anywhere. Do you have plan to release it soon? Thanks!

mcoracin, 2016-06-06:
Hi,
You should be able to use the github version, using the global_conf.json.US902 config file provided in the packet forwarder /cfg directory.
Even though I could not test it myself as I don't have such gateway.

Let me know if you can test it.

Regards

sheenhx, 2016-06-09:
We have successfully build the FPGA version and tested it. And we are the only company which can do the mass production
If you want we can send you one for free:)

mcoracin, 2016-06-09:
Great news, congratulations!
We would be glad to receive one for sure! :)

sheenhx, 2016-06-09:
Please send me your address to xin@airfy.com   :+1:

closed 2016-07-13 by mcoracin