Lora-net/lora_gateway #21: util_pkt_logger fail

verogeorl, 2016-07-27:
Hi,
I tried running the util_pkt_logger with a SX1272 Lora emmitter but the messages aren't written in the log file. 
This is what is in the csv file:
gateway ID,"node MAC","UTC timestamp","us count","frequency","RF chain","RX chain","status","size","modulation","bandwidth","datarate","coderate","RSSI","SNR","payload"
AA555A0000000101,"","2016-07-26 15:16:44.638Z",  68840907, 867600000,1, 0,"CRC_BAD",193,"LORA",125000,"SF7"   ,"4/5",-120,-11.2,"96EE29AF-EDBCB2D4-4F920837-B95474B7-7485BE39-450C3D33-0193FAD0-6FC5A145-B0E50CF4-47394762-A6CB57FD-791D2B4A-ADE6291B-6F48C89A-C89D5275-5056CFC0-3421BBFE-32F61CE1-626CE2DC-6CC7AA3D-694E7300-1D1392CC-DE9D9A3C-B815B48B-33E94415-B4C3EE44-00A88D8A-C6F9FAB5-B274662F-495E6761-BFCEBE0C-BEB0294B-27953B9A-4D3B6F8A-D383AB35-69FE0B05-D696BAEC-7EC45408-76BC0CEA-8BD07ED7-A9F2F137-5EAD1E20-403A824C-60DFBBF0-0A030D7E-D3F3165C-AEFFCAE4-B8E67211-8A"

Can anyone help?
Am I reading the csv file wrong?

closed 2016-08-09 by verogeorl

akm-barnaul, 2017-02-15:
Hi, did you solve this issue?

verogeorl, 2017-02-15:
Yes I did, it was a problem of chanel. I had to chose the same chanel at the sender and receiver side

akm-barnaul, 2017-02-16:
Can you, please, paste here the parts of util_pkt_logger config and node config or source (only the part, that describes chanel selection)?