Lora-net/lora_gateway #26: Running iC880a-SPI with RS Pi3

cattvm, 2016-10-22:
I tried reseting my gateway iC880a-SPI using command **$ sudo ./reset_lgw.sh start** and left the code running for 1 minute. 

how does the actual process starts as soon as the gateway is on:
1. RS Pi 3 ON
2. ./reset_lgw.sh start 
3. ./reset_lgw.sh stop
4. leave for 1 minute
5. Try $ ./util_tx_test -f 868 -r 1257
6. If ok, Then try
7. $ ./poly_pkt_fwd
8. Is that process right.

What are the To-Do if Concentrator doesn't starts "ERROR: [main] failed to start the concentrator"

iBrick, 2016-10-22:
I'd recommend to use the program from IMST documentation to make a reset:
http://www.wireless-solutions.de/images/stories/downloads/Radio%20Modules/iC880A/iC880A-SPI_QuickStartGuide.pdf
see ch. 2.2.2.1.2 Possibility B: Control the GPIO pin via a compiled C-Tool / library

Usually you should not have any troubles with this board even without reset. If you do, first check how you made wiring. If you use jumper wires you should make connection as shorter as possible and you may try to change wires as well. I've noticed that moving from jumper wires to middle board help a lot. And the best is to make a PCB to connect RPi and IMST. 
[image]
.

cattvm, 2016-10-23:
@iBrick  Thanks a lot, was able to get output. But am facing problem with data trans received by gateway. If i check on the Logged data i usually see lots of CRC_ERROR, how to eliminate this error and receive actual data?! Please help me.

[image]

cattvm, 2016-10-23:
Should we have an antenna, I have used a Wire as an antenna, Doesn't it work as an Antenna? My Module is 868Mhz how to make device operate at 868.1/868.2 Mhz
[image]

iBrick, 2016-10-27:
cattvm , github is wrong place for these questions as it doesn't relate to code, you better move to LoRa forums:
1) http://openlora.com/
2) https://www.thethingsnetwork.org/forum/

You may use 8.5 cm wire as antenna.
To make device operate in 868.1 - 3 you need to setup settings in your arduino code and in global_conf.json config file in gateway. 
Regarding CRC_ERROR I have no idea, try to search on forums

mbaldinetti, 2016-11-23:
hi,
in "reset_lgw.sh" must be set "IOT_SK_SX1301_RESET_PIN=17".

did this, i'm able to execute "util_pkt_logger" only with "DEBUG_SPI= 1" otherwise i get "ERROR: [main] failed to start the concentrator".

delay related issue?

vanthome, 2016-12-11:
@iBrick can you share a BOM of your design? I would like to build something similar but I never saw a big prototyping board for the RaspBi... I need to have this :)

iBrick, 2016-12-12:
@vanthome The one on the photo is just simple DIY proto board. If you want I may recommend to order this PCBs [https://pcbs.io/share/rmVdD](url) They are amazing.

vanthome, 2016-12-12:
ok, thx, have you ordered them? I see the min QTY is 4 and I only need one. If you have one left, I would buy it from you ;)

piwi1263, 2016-12-12:
Hi Thomas, since we are in the same country I would buy the other three ...

Champal, 2017-02-28:
Hi, 

I have an RPi3 and IC880a with a custom PCB interface.
Rasbian 8.0
Spi module is spi_bcm2835
lora_gateway version 4.1.2

It's difficult to start packet_fwd or util_tx_test because I have very often "ERROR: failed to start the concentrator"

With DEBUG_REG = 1 and DEBUG_HAL = 1 (lora_gateway/libloragw/library.cfg)
I can see that the init failed with : 
```
lgw_setup_sx125x:407: Note: SX125x #0 version register returned 0x00
lgw_setup_sx125x:415: Note: SX125x #0 clock output disabled
lgw_setup_sx125x:469: Note: SX125x #0 PLL start (attempt 1)
lgw_setup_sx125x:469: Note: SX125x #0 PLL start (attempt 2)
lgw_setup_sx125x:469: Note: SX125x #0 PLL start (attempt 3)
lgw_setup_sx125x:469: Note: SX125x #0 PLL start (attempt 4)
ERROR: FAIL TO LOCK PLL
```
Sometimes it's ok
```
lgw_setup_sx125x:407: Note: SX125x #0 version register returned 0x12
lgw_setup_sx125x:415: Note: SX125x #0 clock output disabled
lgw_setup_sx125x:469: Note: SX125x #0 PLL start (attempt 1)
lgw_setup_sx125x:469: Note: SX125x #0 PLL start (attempt 2)
lgw_setup_sx125x:407: Note: SX125x #1 version register returned 0x21
lgw_setup_sx125x:412: Note: SX125x #1 clock output enabled
lgw_setup_sx125x:469: Note: SX125x #1 PLL start (attempt 1)
lgw_start:788: Note: calibration started (time: 2300 ms)
lgw_start:809: Note: calibration finished (status = 191)
```

As you can see the version register is never the same
0x00, 0x12, 0x21
the good value is 0x21 (http://www.semtech.com/images/datasheet/sx1257.pdf page 27)

The solution for me is to reduce SPI_SPEED in lora_gateway/libloragw/loragw_spi.native.c
```
//#define SPI_SPEED 8000000 // default value, error version register and PLL lock
//#define SPI_SPEED 5000000 // default value, error version register and PLL lock
//#define SPI_SPEED 2000000 // default value, error version register and PLL lock
//#define SPI_SPEED 1500000 // sometimes work, sometimes not
//#define SPI_SPEED 1000000 // ok
//#define SPI_SPEED 800000 // ok but with 50% "version register and PLL lock OK but Calibration failed"
//#define SPI_SPEED 500000 // ok
//#define SPI_SPEED 200000 // ok
#define SPI_SPEED 100000 // ok
```

All test with util_spi_stress are ok at all speed, I try to add an "check version of sx1257 test" but it don't work

With low_speed util_tx_test start each time with right version number but I still loosing packet on my end device ...

Champal, 2017-03-06:
New tests with a modified version of util_spi_stress [util_spi_stress__modif.txt](https://github.com/Lora-net/lora_gateway/files/821630/util_spi_stress__modif.txt)
* Test n°5 with SPI_SPEED in kHz as parameter (-t5 -s8000)
* This test check if the chip version number of the SX1257 n°1 is ok
* 10ms wait between each test and print a '.' or an "!" with the result.
* each 100 tests (1secondes) print a summary

Test under Rasbian 11/01/2017  up to date  (2017-01-11-raspbian-jessie.img)
`Linux rpiLoRa 4.4.48-v7+ #964 SMP Mon Feb 13 16:57:51 GMT 2017 armv7l GNU/Linux`

Under graphical interface (PIXEL) there are lots of error !
nice (sudo nice --20) make no change
* 8Mhz : 75% error
* 1MHz : 15-30% error, mouse activity increase error rate, 60% error while playing a video on youtube (20% CPU)
* 100kHz : 0% error even with mouse, video, ...

After reboot without X 
* 8MHz : 0% error ! but if I switch to another terminal (CTRL+ALT+Fn) => error !
* 1MHz : less sensitive, but error when it's the first time I switch to a new Fn

After reboot, without screen , only ssh
* 8MHz : 0% error with 0 activity but for example open another ssh, ls-lR, make errors !
* 1MHz : better but not perfect
* 100kHz : rock stable (ls -LR /) without renice. Even with a `sysbench --num-threads=4 --test=cpu --cpu-max-prime=4000 --validate run` 

After all this I make test with an oscilloscope and there is a surprise ! 
Test : 
- one remote ssh with "top"
- one remote ssh for `sysbench --num-threads=4 --test=cpu --cpu-max-prime=6000 --validate run`  the test make 22s to complete
- one remote ssh with `util_spi_stress -t5 -s <spi_speed_in_kHz>

Results : 
* 790 kHz: real speed 492kHz, when I run sysbench => 787kHz and no error 
* 791 kHz : real speed 494kHz, when I run sysbench => 790kHz and error 
In fact the system speed change `sudo cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq` give 600000 without sysbench and 1200000 during sysbench
* 790kHz : work even with `renice 19`
* 791kHz : `renice 19` make more errors but no benefits with `renice -20` 

Last test for today : Raspbian Lite 18/03/2016 (2016-03-18-raspbian-jessie-lite.img)
`Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux`

* 8MHZ : error during sysbench
* 4MHz : error during sysbench
* 3MHz : same problem as 2MHZ
* 2MHz : real freq 1.25MHz and 2MHz during sysbench. No error during sysbench but errrors just after (from the end of test to the return @1.25MHz)
* 1.550MHz : same problem as 2MHZ
* 1.5MHz : rock stable (932kHZ/1.492MHz)

Raspbian releases notes :
https://downloads.raspberrypi.org/raspbian_lite/release_notes.txt

ckt1010, 2017-03-15:
Hi @Champal ,
Thanks for your test! I have same problem! And your test is really shock me.
Is that means we can't use speed higher than 100kHz? And I wonder if it is spi driver's problem, because nice is useless.
So is there any other good hardware platform? If we can't fix this problem on Pi.

mogyoros, 2017-03-23:
Dear @Champal !
Thumbs up for your findings! I saw the exactly same issue happening on my RPI3 + shield + iC880A.
After setting the SPI speed to 1MHz, it is working solid rock!
Thanks for sharing this mate!

_On the other side, I was wondering, if this behavior shouldn't be related to - somekindof - linux kernel change that affects the kernel SPI driver?_

Regards,
tamas

Yaoshicn, 2017-03-29:
Hi, @mogyoros,

I am wondering which type power supply unit you are using? 

I tried 5V 2.5A, which in the default power supple unit for Rpi 3, for my Rpi3 + shield + iC880A. It came with the error "failed to start the concentrator". The green LED on iC880A is on. My idea is that maybe the power is not sufficient.

Thanks in advance.

Regards,
Xuanliang

mogyoros, 2017-03-29:
Dear @Yaoshicn !

Well, for the first try I was using a 5V/2A USB power supply, which is a tablet charger. Then after seeing this random behavior, I started reading through different forums and the specs of RPI3 and iC880A. Several topics advised using an external power supply to be used, since through the onboard USB connection of RPI3 sometimes can not be supply the necessary power to external devices.
I moved to a switching power supply adapter of 5V/2A, and powered the iC880A and the RPI3 via the shield. Before that I measured the power supply how much voltage it outputs. Before connecting 5.11V, after connecting to the shield it dropped to 5.07V. But this change also did not solved the issue, so I swapped the external switching power supply adapter to an another 5V/2A version. Same: no change, sometimes radio module starts, sometimes not.
So I have enabled the DEBUG_REG before recompile and saw the above debug messages about PLL start errors. Then googled it, and found this topic, which led me to the solution to modify the SPI speed to 1MHz. 
Since then, no problems is seen, whatever power supply is used (via RPI3 onboard USB connector, or external)

Regards,
Tamas

Yaoshicn, 2017-03-29:
Dear @mogyoros ,

Thank you for your help and patience. I solved my problem!

Best Regards,
Xuanliang

closed 2017-03-31 by mcoracin

vitomakes, 2017-05-30:
Thanks for the suggestions, same problem here. Stupid question, where do I change the SPI speed?

pat1, 2018-07-06:
I am running some test with Raspberry 3B+ and centos 7 distribution:
Kernel: Linux version 4.9.80-v7.1.el7 (mockbuild@armhfp-01.bsys.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-16) (GCC) ) 
I run the test with: 
  sysbench --num-threads=4 --test=cpu --cpu-max-prime=6000 --validate run
running with execution time (avg/stddev):   45.1321/0.02 and %CPU at 398 for sysbench (from top).
I run util_spi_stress with all the test number without problems at 8Mhz and 1Mhz.

Instead I had a lot of problems with radio interference due to nearby power transmitters; I therefore suggest that you consider the length and type of the SPI's hardware connection if you are in an environment with considerable electrical noise.