Lora-net/lora_gateway #27: util_tx_text

mahendratallam, 2016-10-27:
I am trying to test Util_tx_test present in the Lora net folder. Steps followed
1. Compiled the Util_tx_test module and generated the executable.
2. Run the program with frequency setting to 867.9 and radio type as 1257.
3. Able to see the packet transmission successful but not able to receive it.
4. Using GL6509 Low Power WAN module to receive the data. Able to receive the data packet by the util_pkt_logger sending by GL6509 though.

Please find the attached screenshot. 
[image]
Please help me if i miss anything.

iBrick, 2016-10-27:
If you receive it in pkt_logger that means that packet is OK. What is missing is decoding AES message from payload. Where are you going to receive packet? you need LoRa server for this.

mahendratallam, 2016-10-27:
I am trying to receive the Lora packet through  "GL6509 Low Power WAN module". 
Through the serial command "AT+DRX?.  i am not able receive through this.
Packet transmission seems fine as showing in screen shot.
please say if i miss anything.

iBrick, 2016-10-27:
Ok, I missed the point. Now I've checked what is GL6509. It is a simple LoRaWAN class A node. 
You are now here in lora_gateway repository. This code won't work with your node at all! GL6509 is not a gateway, it's just an end node.
Looks like you need to start from the basics or I miss something.
You need 'Sensor' -> 'Gateway' -> 'Server' architecture. Google for tutorials.
As this topic doesn't relate to code I recommend to close it. You better move to LoRa forums:
1) http://openlora.com/
2) https://www.thethingsnetwork.org/forum/

mahendratallam, 2016-10-27:
sorry to bother you..
I am sending the packet using Lora CONCENTRATOR (Gateway) SX1257 and trying to receive it using the node GL6509. The code used to transmit is util_tx_test.

iBrick, 2016-10-27:
You won't be able to use util_tx_test to send data to your node. 
You need LoRa server to do this. Util_tx_test doesn't have even functions for encoding payload with NWKSKEY and APPSKEY: https://github.com/Lora-net/packet_forwarder/blob/master/util_tx_test/src/util_tx_test.c
util_tx_test is quite useless program.

mahendratallam, 2016-10-27:
Thank you for your inputs..

closed 2016-10-27 by mahendratallam