Lora-net/lora_gateway #28: reference design doc?

laoqr, 2016-10-27:
Hello,
I am wondering if there are some reference design documents for lora-gateway with sx1301?

Thanks!

iBrick, 2016-10-27:
No working, only this: https://github.com/will127534/LoRa-concentrator

Jungok, 2016-10-28:
https://github.com/Lora-net/lora_gateway
This source is contiributed by semtech

zhoubx123, 2016-12-01:
Jungok，can you tell me your design file with the fpga?Raspberry pi  board needs to connect what pins?
thank you !

Jungok, 2016-12-12:
@zhoubx123 
Hello, Zhoubx123. I am having the IOT SX1301 starter kit with Raspberry pi 2 for Korea. The board name is PCB_E3401.5. It simply connect to Raspberry through SPI. I couldn't share my design file because it is confidential. However, You can get it from Semtech.  

In addition, I heard that FPGA is only for LBT, and some countries don't need to LBT.

zhoubx123, 2016-12-12:
@Jungok

zhoubx123, 2016-12-12:
@Jungok

zhoubx123, 2016-12-12:
@Jungok

zhoubx123, 2016-12-12:
@Jungok

zhoubx123, 2016-12-12:
@Jungok
Can you tell me your email ?
I have some questions to ask you.
My email is 724784058@qq.com.

zhoubx123, 2016-12-12:
Jungok
Can you tell me your email ?
I have some questions to ask you.
My email is 724784058@qq.com.

ucasws, 2017-03-15:
@Jungok  could you tell me how to sign NDA with semtech? please contact me,thank you ,my eamil is shuai.wang@mail.sim.ac.cn

mcoracin, 2017-03-31:
Please contact Semtech support for this kind of requests.

closed 2017-03-31 by mcoracin