Lora-net/lora_gateway #29: Receive unfiltered LoRa Frames

Smarco10, 2016-11-02:
Hi All,

I'm currently working on a projet where I have to receive LoRa frames on a gateway.

I'm facing a trouble receiving frame which come from other GW. I guess that the SX1301 is defaulty configured to filter on end-devices uplink messages.

Is it possible to bypass this filter (through SPI regs by example) in order to receive all kind of messages?

Thank you for your assistance.
Sincerely

nestorayuso, 2016-11-02:
Hi,
gateway -> node data has the I/Q inverted compared to node -> gateway data.

I think you can configure SX1301 with I/Q inversion, in this case you will receive data from other gateways but will not receive from nodes.

Smarco10, 2016-11-02:
Thank you for your quick reply.
I've seen that seems possible on the SX1272 (node-side module) but I didn't see any register with this desciption. Do you have any clue about that?

nestorayuso, 2016-11-02:
check this registers:
LGW_RX_INVERT_IQ
LGW_MODEM_INVERT_IQ
LGW_MBWSSF_MODEM_INVERT_IQ

https://github.com/Lora-net/lora_gateway/search?utf8=%E2%9C%93&q=LGW_MODEM_INVERT_IQ

Smarco10, 2016-11-02:
OK thanks.
Do you have any documentation on I/Q setup?

mcoracin, 2016-11-09:
Hello,

You cannot configure a Gateway to receive packets from both nodes and other gateways.
But as said above, if you want to  configure it to receive frames from other gateways, you have to set this:
               lgw_reg_w(LGW_MODEM_INVERT_IQ,0);

closed 2016-11-09 by mcoracin

davidonet, 2017-01-25:
Tried it on multitech conduit but get only up message.

mcoracin, 2017-01-25:
In order to receive frames on a gateway coming from another gateway, you need to set:
lgw_reg_w(LGW_MODEM_INVERT_IQ,0);
lgw_reg_w(LGW_ONLY_CRC_EN,0);

davidonet, 2017-01-27:
Doesn't work can't see any down packet and keep receiving up packet...
Anyone succeed in this ?

Smarco10, 2017-01-27:
I got the same issue 2 months ago, but I didn't know that there where a LGW_ONLY_CRC_EN reg.
I solved the problem with the second line and it work really well.
Verify that you are not to close (>10m) to the Gateway which may have a positive-gain antenna.
Did you you filtered on the packet status? Check that you allow STAT_NO_CRC.

mcoracin, 2017-01-27:
I'm using a gateway to simulate a mote both for RX and TX with attached patch applied on the HAL.
The attached patch is required for RX.
For TX, just have to set txpkt.invert_pol to false.

Smarco10, 2017-01-27:
And do not forget to set no_crc to true if you generate RX from a GW.

davidonet, 2017-01-30:
Where can I find the patch required for RX ?

davidonet, 2017-02-03:
Tried it on imst rpi and keep on receiving up message and no down :-(

Hexcodes, 2019-05-20:
Did anyone find a solution for this? Currently I'm stuck at the same problem like @davidonet where the Gateway detects only some packets, but not all.