Lora-net/lora_gateway #30: RF module sometimes shut down itself

joyzhang15, 2016-11-09:
I have some gateways, sometimes RF module can't receive message from node without any error raised. And if I restart the application or restart the RF module, it works fine again. I have RF modules from several manufacturer(including IMST) and many of them have the same problem. Have you ever saw this problem?

mcoracin, 2016-11-09:
What could happen is a wrong SPI transfer from host to RF module which leads putting the SX1301 or radio in a wrong state.
As there is no CRC on SPI transfer, if somehting get corrupted during the transfer, for example the register address, or a read becoming a write, it could have a fatal effect.

It is recommended, on the top application, to add a watchdog to detect such cases, and trigger a restart.

closed 2016-11-09 by mcoracin

joyzhang15, 2016-11-10:
Thanks for your reply. I was wondering if the problem is caused by the hardware or it is caused by SPI transfer. And I was wondering if your gateway also have the same problem?

mcoracin, 2016-11-10:
I don't know for your particular case.
Just we saw this kind of problem once, and the issue was due to SPI transfer. It can depend a lot of the type of connection you have between the host and the radio module, how it is wired.

I never faced the issue myself, but we know it can happen.

How often is it on your side? If it is every 5 minutes, there is a big problem. If it happens after a week or so, the watchdog is definitely the way to go.