Lora-net/lora_gateway #31: The latest version of the LBT gateway function, raspberry and SX1301 concentrator pin how to connect?

yanzi418, 2016-12-01:
I would like to know, the latest version of the gateway connection diagram? Sx1301 how to connect with raspberry's SPI ? FPGA how to start? After the operation of the gateway, prompted FPGA did not start . lgw_connect:524: INFO: no FPGA detected or version not supported (v0)
I don't know how to connect sx1301 to drive the raspberry pie FPGA (control sx1272) and sx1255

yanzi418, 2016-12-06:
ice40hx1k和ice40lp1k能够公用SX1301_FPGA_125K_NOTCH_LBT_bitmap_v27.bin相同程序吗
ice40hx1k和ice40lp1k能够公用SX1301_FPGA_125K_NOTCH_LBT_bitmap_v27.bin相同程序吗
Can ice40hx1k and SX1301_FPGA_125K_NOTCH_LBT_bitmap_v27.bin share the same program as ice40lp1k

mcoracin, 2016-12-09:
Check the packet forwarder Wiki page for having more info about how to connect a rapsberry pi to the gateway board:
https://github.com/Lora-net/packet_forwarder/wiki/Use-with-Raspberry-Pi

There is nothing to do to start the FPGA, it will boot at power on.
If it is well programmed, and the SPI connexion is ok, the HAL should detect it.

closed 2016-12-09 by mcoracin