Lora-net/lora_gateway #32: FPGA not start!

zhoubx123, 2016-12-01:
I use raspberry pi and SX1301AP2-PCB_E336, I have use raspberry pi pins MOSI , MISO , SCLK , CE0 and CE1 . CE0 choice SX1301 , and CE1 choice FPGA. When I start getway and module，getway can receive Join request , but mudule cannot receive Join Accept. If I connect SPI error , or need updata SPI drive ?

mcoracin, 2016-12-09:
Hello,
Which FPGA image have you programmed?
I don't understand why you say "and CE1 choice FPGA"?

To plug a SX1301AP2-PCB_E336 to a raspberry pi, you just have to connect CE0 to the pin 1 of the GW board.
Before testing with a module, have you checked that your GW is functionnal (SPI OK, packet forwarder starting with no error etc...?)

You can use the util_spi_stress to check the SPI connexion.

zhoubx123, 2016-12-10:
Thank you very much.
I must load SX1301_FPGA_125K_NOTCH_LBT_bitmap_v27.bin before use FPGA，Or the gateway is automatically loaded at boot？
Is after load FPGA image, and then connect a raspberry pi MOSI , MISO , SCLK , CE0.
just do like this is OK？

mcoracin, 2016-12-12:
Hello,
You must program the FPGA before. I do it using a JTAG and Lattice Diamond Programmer software.
Once it is programmed, as you say, you connect your raspberry pi as described here:
https://github.com/Lora-net/packet_forwarder/wiki/Use-with-Raspberry-Pi

zhoubx123, 2016-12-14:
FPGA is broken, can I use ICE40HX1KVQ100 instead , I just have this.

closed 2017-03-31 by mcoracin