Lora-net/lora_gateway #38: FPGA hex format

linux4life798, 2017-01-19:
I am trying to flash the FPGA using the SEGGER JFlashSPI utility, but it seems to want a binary image.

I see that the FPGA images are now in a .hex format. What type of hex format is this? Intel, Motorola, Extended Tektronix hex?
Can I simply build a binary image composed of the bytes spelled out in the hex file?

Thanks,

Craig

linux4life798, 2017-01-19:
I figured it out. Here is what I learned:

The Segger seems to be useless without proper SPI licensing and a more expensive programmer.
You can program the FPGA's flash using a Raspberry Pi, flashrom utility, and a custom hex to bin script that I will attach here.
Yes, the hex file seems to just be a list of the bytes that compose the FPGA flash image.

1) Use my attached shell script that takes Semtech's published FPGA ".hex" file and produces a proper sized ".bin" file for flashrom. The script simply concatenates the bytes with extra zero padding.
EX: `./sx1301hex2bin.sh ~/src/lora_gateway/fpga/SX1301_FPGA_NOTCH_PROG_SPECTRAL_SCAN_v31.hex ~/newimage.bin`
2) Connect the RPi SPI0 w/CE0 to the SX1301AP2 FPGA SPI program port and enable SPI using the raspi-config tool.
3) Install flashrom
EX: `sudo apt-get install flashrom`
4) Flash the FPGA using flashrom
`flashrom -p linux_spi:dev=/dev/spidev0.0 -w ~/newimage.bin -V -c "M25P10-A"`

You may want to backup the original image by doing the following:
`flashrom -p linux_spi:dev=/dev/spidev0.0 -r ~/original_image.bin -V -c "M25P10-A"`

Best,

Craig

[sx1301hex2bin.sh.zip](https://github.com/Lora-net/lora_gateway/files/715712/sx1301hex2bin.sh.zip)

mcoracin, 2017-01-26:
Thank you Craig, it will be very useful for a lot of users I think!

pierreduf, 2017-07-31:
Hi Craig,

Thank you for that piece of valuable information ! I'm trying to backup (and change) the FPGA firmware on a LoRa GW but that doesn't seem to work.

I've compile flashrom for the right architecture (and indeed it's starting and working without any error) but it doesn't seem to find any matching FPGA (even forcing the M25P10 like you did).

My SPI device is /dev/spidev1.0, and I've adapted the command accordingly.

Do you have any other tips or experience on other GW that I could try ? I guess that's related to the SPI muxing done by the SX1301 : I've got the feeling that flashrom sees only the SX1301 and not the FPGA behind it (but I may be wrong, I'm not really an expert of that kind of thing).

BR,

Pierre

linux4life798, 2017-07-31:
Hi Pierre,

You need to make sure that you are connected to the actual flash SPI channel and not the 1301 / FPGA normal data SPI channel. Note that the M25P10 device is actually just a standard SPI flash chip that is then connected to the iCE40 FPGA to load the image at boot.

For future reference, I believe the FPGA serves as the SPI mux between the SX1301 and the extra SX1273/76, but of course this is only applicable to the normal data SPI lines.

linux4life798, 2017-07-31:
If I recall correctly, you also need to hold the board in reset so that you can talk to the flash chip directly.

pierreduf, 2017-07-31:
Thank you Craig for you quick answer ! I think that's clearer for me. I'm going to try a few things and let you know if I succeed.

linux4life798, 2017-08-02:
Sounds good, best of luck!

pierreduf, 2017-08-07:
Done it with a Raspberry Pi and another tool called spiflash and it worked ! Thank you for your help.

closed 2021-06-24 by rbaldwin13