Lora-net/lora_gateway #42: Use interrupt mode

x893, 2017-02-26:
Hi,
possible change software (add #define ... to library.cfg for example) so i can use GPIO mode 2 and GPIO0 as "rx_buffer_not_empty" on PIN7 to implement interrupt mode on RPi ?

Thanks in advance
Mitek

mcoracin, 2017-03-31:
Hi,
The SX1301 does not send any signal/interrupt when a buffer has been received, the host has to poll.
Best regards

mcoracin, 2017-03-31:
Hi,
We don't want to support this configuration, but if you want to do it, you can find some information here:
https://github.com/Lora-net/lora_gateway/wiki/SX1301-GPIO-output-configuration

closed 2017-03-31 by mcoracin

x893, 2017-03-31:
Thanks !
i use iC880A version 1.0 with STM32 board (not RPi) and easy implement interrupt mode.

mcoracin, 2017-04-03:
Nice, is your STM32 code for the HAL and packet forwarder available somewhere?

x893, 2017-04-03:
yes, i use HAL but add linux emulate library (tread, time, etc.)
Now i use 32F746GDISCOVERY but possible F4xx - depence from network traffic.
ESP8266 or GSM module as uplink to network server.
Now in test stage.

mcoracin, 2017-04-03:
Are you using any kind of operating system on the STM32 to handle threads for the packet forwarder?

If you can share the results of you test at some point, it could be interesting. :)

x893, 2017-04-03:
Yes, i use FreeRTOS but decorate it for linux calls so not need change your packet_forwarder source code.

mcoracin, 2017-04-04:
ok, thanks for the info

nestorayuso, 2017-04-10:
Nice to see a new version, PicoGW with STM32 implementing USB interface to SX1308

x893, 2017-04-15:
PicoGW look strange - why need 2 processors STM32 and RasPi ?
I'm very inetersted in SX1308 PCB but i can't find a seller.
I use SX1301 version with STM32 (minimum flash 128K because packet_forwarder need ~100K) and ESP8266 (WiFi), LWIP (Ethernet) or with cheapest GSM modem (most applications).

mcoracin, 2017-04-18:
The goal of this PicoGW is to have a LoRa Gateway USB dongle which can be connected to a RPi, a PC, or any host with a USB port.
The purpose of the STM32 here is just to handle the communication through USB to the SX1308, and also, reduce the power consumption of the SX1308 to match with USB port constraints.
The packet forwarder still remains on the HOST, whatever it is.

The goal was not, yet, to have a full gateway running on the STM32.

x893, 2018-08-31:
What's a problem ? You can contact via skype. But now i've only RAK831 board. I port to STM32 (long time ago) only for tests interrupt mode from SX1301.

hoangtrung101, 2018-09-12:
@x893 I use RAK831 with STM32f103VE. The purpose is that I want to understand the communication process between RAK831 and PI3. I porting library of SX1301 from https://github.com/Lora-net/picoGW_hal/tree/master/libloragw to stm32 but when I test example like test_loragw_reg.c the receiver data is wrong. here is my code https://github.com/hoangtrung101/RAK831_STM32. 
I tried test send/receiver data between STM32 and RasPi by running the file  "test_loragw_spi.c" and it works correctly

x893, 2018-09-21:
I think SPI problems can easy resolve with debugger.