Lora-net/lora_gateway #48: check duplicated packets

mikumaycry, 2017-04-10:
I came across the as problem in issue below:
https://github.com/brocaar/loraserver/issues/159
The node sent first packet to gateway, gateway received packet but didn't response to it.
So the node sent same data again but used a different frequency and expected a different receive frequency.And the server handled first packet and drop the latter one.
Although gateway received first packet response, data couldn't be transferred to node.
I wonder is it better to check duplication on gateway or on server?

mcoracin, 2017-04-10:
Hello,
As suggested in the issue you mentioned, the root issue of duplication is certainly due to the fact that the node is too close from the gateway.

The server should detect the packet duplication and keep the one with the best RSSI.

Is it the case?

mikumaycry, 2017-04-11:
Thank you for your reply.
Maybe I just got into the wrong way. The gateway is not responsible for handling duplication.

In my test, server always responses to the first packet and drops the rest, ignoring RSSI and SNR.
But node expects the last packet's downlink frequency.

I gonna try decrypting every packet to check duplicated data and response to the last packet.

mcoracin, 2017-04-11:
You should rely on the RSSI of the packet, it's not a matter of which packet is first or last I think.

closed 2017-09-21 by mcoracin