Lora-net/lora_gateway #49: Raspberry Pi  to ic880A-spi communication fail at 8Mhz

Richaagarwalrn, 2017-04-14:
I am trying to communicate Raspberry Pi board to ic880A-spi over spi .At spi speed 1 MHz its working fine but at 8 MHz sometime its working or sometime not due to PLL lock lose

 do {
            if (cpt_attempts >= PLL_LOCK_MAX_ATTEMPTS) {
                DEBUG_MSG("ERROR: FAIL TO LOCK PLL\n");
                return -1;
            }
            sx125x_write(rf_chain, 0x00, 1); /* enable Xtal oscillator */
            sx125x_write(rf_chain, 0x00, 3); /* Enable RX (PLL+FE) */
            ++cpt_attempts;
            DEBUG_PRINTF("Note: SX125x #%d PLL start (attempt %d)\n", rf_chain, cpt_attempts);
            wait_ms(1);
        } while((sx125x_read(rf_chain, 0x11) & 0x02) == 0);
From these part of code it get return whenever "failed to start concentrator". Please suggest how to remove this error.

Champal, 2017-04-19:
Hi, 
more test here https://github.com/Lora-net/lora_gateway/issues/26#issuecomment-283089276
short : depends on kernel version, no real explication

mariosmail, 2017-07-13:
Similar problem here but in my situation I can see communication through SPI with logic analyzer running the "test_loragw_spi" but I got Always "ERROR: failed to start the concentrator" when i run "util_pkt_logger"

Furthermore when I try to run just the "test_loragw_reg" I got a lot of register "MISMATCH".
So SPI does't always work for me.

mcoracin, 2017-07-17:
Hello,

The "FAIL TO LOCK PLL" is often due to a weak power supply. How do you power up the SX1301 board?
Through the RPi?

How are the status LED of the Raspberry Pi? Is the red led blinking from time to time?

In order to have a stable boot-up, you need to check that the following points are ok:
- a stable power supply. 1A for the SX1301 board alone, or 2A for RPi+SX1301 board. The official RPi power supply is ok.
- a stable SPI link (checked fro hours using util_spi_stress (not test_loragw_spi).
- the SX1301 to be put out of reset through the dedicated pin.

mariosmail, 2017-07-20:
Hi, 
First of all the SX1301 is powered up from the RPi.
On the other hand let me tell you that the TTN-gateway firmware that use also same repository make my system work and even if I run tests firmware from the already installed "lora-gateway" repository its everything working.

So after this observation I tried to clone the repositories to the /opt folder of my totally clean system, Build the whole repository and after that I tried to run again one-by-one every test on "libloragw" folder. Finally once again I found out that my tests don't run properly.

Mkatari2293, 2017-08-29:
Hi Mario

Can I have a look at your SPI signal communication on Logic Analyzer. I am going through the same as you do. I could be able to see the SPI communication through Logic Analyzer. But getting an "Error: Concentrator Unconnected". Couldn't able to read the Version ID in kernel.

Will be waiting for your response.

Thanks

embed-engineer, 2021-05-04:
If you have a stable SPI connection (spi stress is good) and even you get PLL lock problem in 8MHZ SPI speed, maybe you can try my advice..

I added a 1 micro second sleep in lgw_reg_w() function as below.(in libloragw/src/loragw_reg.c)

But you need to write a wait_us() function in loragw_aux.c.

I was getting PLL LOCK fault above 3MHz SPI speed and It works for me well at 8MHz now.

If people want, i can send pull request also with corrections...

```
/* Write to a register addressed by name */
int lgw_reg_w(uint16_t register_id, int32_t reg_value) {
    int spi_stat = LGW_SPI_SUCCESS;
    struct lgw_reg_s r;
  
    wait_us(1);
    
    /* check input parameters */
    if (register_id >= LGW_TOTALREGS) {
        DEBUG_MSG("ERROR: REGISTER NUMBER OUT OF DEFINED RANGE\n");
        return LGW_REG_ERROR;
    }
```

closed 2022-10-05 by smtc-bot