Lora-net/lora_gateway #50: util_pkt_logger: ERROR failed to load fw 1?

CanhThang, 2017-04-22:
I'm using Raspberry Pi 3, RHF0M301 - 915 to make a gateway. I ran util_pkt_logger and got this:

ERROR: failed to load fw 1
ERROR: Version of calibration firmware not expected, actual:0 expected:2

Can anyone please check and help me? I am new to github.

reissjason, 2017-04-24:
Check the supplied power. If the read/write is failing the read will return 0.

ClementHuchet, 2017-04-28:
Hi, verify you wiring and make sure you connected the grounds of both devices.

CanhThang, 2017-05-06:
Thank you, I didn't know that I need to connect the grounds of both devices. It worked.

closed 2017-05-06 by CanhThang

ClementHuchet, 2017-05-06:
My pleasure , cheers ;)

van-cuong-nguyen, 2017-11-18:
Dear @CanhThang,
Could you share with me your experience in building a gateway with RHF0M301 - 915  and Pi 3?
I got this error. Many thanks.
[image]

3nd0y, 2018-01-08:
Hi i have the same issue, i checked the wiring is good, then ground both of them connected but still have error
ERROR: failed to load fw 1
ERROR: Version of calibration firmware not expected, actual:0 expected:2

is there other answer?

arihantdaga, 2018-03-16:
Same here with me. My wirings are good. I bought this lite Gateway which comes pre assembled, i dont think wiring is having anything wrong, i am giving power of 5v 2A. Still getting this same error.. 
ERROR: failed to load fw 1
ERROR: Version of calibration firmware not expected, actual:0 expected:2

And using util_spi_stress -t 4
Returns read values 00
Can somebody please suggest what is going wrong here..

arihantdaga, 2018-08-04:
I am facing the same issue.. 
Did anyone figure it out..

arihantdaga, 2018-08-04:
@3nd0y