Lora-net/lora_gateway #51: loragw_radio: SX125x #0 PLL??

CanhThang, 2017-05-18:
After reseting by ./reset_lgw.sh, I ran ./util_pkt_logger, and the code can only start the concentrator for 1 out of 5 or 6 times. I found that the problem is from line 460 in loragw_radio.c, where start and PLL lock.
[image]
Can anyone tell me what is this problem with PLL, because it always wrong and I can rarely start the concentrator.
Thank you in advance.

mcoracin, 2017-05-22:
Which kind of power supply are you using?

I've seen this problem very often due to weak power supply.

Regards

CanhThang, 2017-05-24:
I use an external power supply with the voltage of +5V. Is there any requirement with the supply current??

mcoracin, 2017-05-24:
Yes, you need a power supply that can deliver 1A.

CanhThang, 2017-05-24:
I tried, but it didn't work. Refered to https://github.com/Lora-net/lora_gateway/issues/26#issuecomment-283089276, I reduced the spi speed and it worked, but there are maybe chances losing packet.

closed 2017-05-24 by CanhThang