Lora-net/lora_gateway #52: Requirements to receive (or send) packet?

CanhThang, 2017-05-25:
I use Raspberry Pi 3 with RH0M301 (from RisingHF) to make a LoRa gateway. I try util_pkt_logger to receive packet from my end devices, with config parameter from here: https://github.com/TheThingsNetwork/gateway-conf/blob/master/AS2-global_conf.json
I found out that sometimes my gateway receive a packet from unknown device, not from my devices. Can anyone tell me what are the requirements which both gateway and device have to follow to send or receive packet?
Thank you in advance!

mcoracin, 2017-06-06:
Hello,

Have you progressed on this?

Do you have access to a reference gateway or a reference node that is proven to be working, in order to separate issues?

Which software are you using on the end-node side?

Maybe one first thing to check is that you are using the same syncword on both ends:
- On the GW side, the "lorawan_public" field set to true indicates that you are using a syncword of 0x34 (instead of 0x12).
- On the node side, by default, the register RegSyncWord is sent to 0x12, unless your sw sets it to 0x34.

CanhThang, 2017-06-07:
It's true. I should have checked the syncword first.
I have another question about FPGA. I use RH0M301 (from RisingHF) and I can not start FPGA. Can you check for me if this device supports FPGA?

mcoracin, 2017-06-07:
I don't think there is a FPGA on this board, but better check with RisingHF. I nerver used this board.

CanhThang, 2017-06-08:
So that I can not use LBT with this board.
1. How can I use LBT without FPGA?
2. If question number 1 can not be accomplished ,what should I do to send packet by lora_gateway code without LBT? Should I just disable LBT functions and then use lgw_send?
Thank you so much @mcoracin

mcoracin, 2017-06-08:
You cannot do LBT without the FPGA and the extra SX1272 which is there for carrier sensing while the gateway is operating.

In this case, you have to disable the LBT feature in the hal/packet_forwarder.

CanhThang, 2017-06-08:
I now can receive packet from my node (RF95). I tried to send packet from RH0M301 back to node by util_tx_test. I saw the successful packet transmission but my node did not receive the packet. Can you tell me what is missing here?

mcoracin, 2017-06-08:
Can you give the complete command line you are using with util_tx_test?
Which software do you use on the end-node side?

CanhThang, 2017-06-09:
Sorry I use util_tx_continuous. I mix it with util_pkt_logger to create a main file like this:
`/*
 / _____)             _              | |
( (____  _____ ____ _| |_ _____  ____| |__
 \____ \| ___ |    (_   _) ___ |/ ___)  _ \
 _____) ) ____| | | || |_| ____( (___| | | |
(______/|_____)_|_|_| \__)_____)\____)_| |_|
  (C)2013 Semtech-Cycleo
Description:
    Configure LoRa concentrator and record received packets in a log file
License: Revised BSD License, see LICENSE.TXT file include in the project
Maintainer: Sylvain Miermont
*/

/* -------------------------------------------------------------------------- */
/* --- DEPENDANCIES --------------------------------------------------------- */

/* fix an issue between POSIX and C99 */
#if __STDC_VERSION__ >= 199901L
    #define _XOPEN_SOURCE 600
#else
    #define _XOPEN_SOURCE 500
#endif

#include <stdint.h>     /* C99 types */
#include <stdbool.h>    /* bool type */
#include <stdio.h>      /* printf fprintf sprintf fopen fputs */

#include <string.h>     /* memset */
#include <signal.h>     /* sigaction */
#include <time.h>       /* time clock_gettime strftime gmtime clock_nanosleep*/
#include <unistd.h>     /* getopt access */
#include <stdlib.h>     /* atoi */
#include <wiringPi.h>	/* for Raspberry Pi*/

#include "parson.h"		/* convert json file */
#include "loragw_hal.h"
#include "LoRaMAC.h"
#include "main.h"

/* -------------------------------------------------------------------------- */
/* --- PRIVATE MACROS ------------------------------------------------------- */

#define ARRAY_SIZE(a)   (sizeof(a) / sizeof((a)[0]))
#define MSG(args...)    fprintf(stderr, args) /* message that is destined to the user */

#define TX_RF_CHAIN             0    /* TX only supported on radio A */
#define DEFAULT_RSSI_OFFSET     0.0
#define DEFAULT_FREQ_HZ         923800000
#define DEFAULT_DIGITAL_GAIN    0
#define DEFAULT_DAC_GAIN        3
#define DEFAULT_MIXER_GAIN      14
#define DEFAULT_PA_GAIN         3
#define DEFAULT_MODULATION      "LORA"
#define DEFAULT_SF              8
#define DEFAULT_BW_KHZ          125
#define DEFAULT_BR_KBPS         50
#define DEFAULT_FDEV_KHZ        25
#define DEFAULT_BT              2
#define DEFAULT_NOTCH_FREQ      129000U

/* -------------------------------------------------------------------------- */
/* --- PRIVATE VARIABLES (GLOBAL) ------------------------------------------- */

/* configuration variables needed by the application  */
uint64_t lgwm = 0; /* LoRa gateway MAC address */
char lgwm_str[17];

/* clock and log file management */
time_t now_time;
[... 405 lines cut ...]
        lgw_reg_w(LGW_TX_OFFSET_I, 0);
        lgw_reg_w(LGW_TX_OFFSET_Q, 0);
    }
    
    uint16_t cycle_count = 0;
    
    /* TX Continuous */
    while(1)
    {
		/* Send packet */
        printf("Sending packet number %u ...\n", ++cycle_count);
        i = lgw_send(txpkt);
		if(i == LGW_HAL_SUCCESS){
			printf("Packet Sent\n");
			printf("-----------\n");
		}
        wait_ms(2000);
	}
	
    MSG("INFO: Exiting packet logger program\n");
    return EXIT_SUCCESS;
}

/* --- EOF ------------------------------------------------------------------ */`
I use Keil C and a Cortex - M0 MCU.

mcoracin, 2017-06-09:
why don't you use the packet forwarder to send and receive packets?
[https://github.com/Lora-net/packet_forwarder](https://github.com/Lora-net/packet_forwarder)

On the end-node side, which LoRaWAN software are you using?

I don't understand the point of mixing util_pkt_logger with util_tx_continuous...

CanhThang, 2017-06-09:
I want to make my own gateway code from lora_gateway code, and connect to my server by MQTT. I changed the gateway module recently from SX1276 (small project) to RH0M301 (for bigger project). Now I am trying to send and receive packet from gateway to node, the other parts I have done before.

mcoracin, 2017-06-09:
You could just replace the UDP part of the packet_forwarder with MQTT, right?
util_tx_continuous is more a RF validation tool, rather that something to send LoRa packets.

If you want to get support, I would suggest to validate that you gateway is correctly working using a proven software, with a working LoRaWAN end-node.
Then it is up to you to change everything with what you want. :)

CanhThang, 2017-06-10:
Hmm I am using non protocol packet. I tried your code: https://github.com/Lora-net/lora_gateway/blob/master/util_tx_test/src/util_tx_test.c to send packet (no mix with anything) in the same parameters. Are there any requirements with antenna, or syncword, or anything that I forget to set the same in gateway and node?

mcoracin, 2017-06-14:
The polarity has to be inverted, you need to add the -i option to util_tx_test.

CanhThang, 2017-07-13:
Dear Mr @mcoracin,
I now can control SX1301 and do the tasks I want. Thank you for supporting me devotedly and enthusiastically.

I still have some problems that need you to help:

1. About the stability (or duration - sorry I don't know the specific technical word) of network:
If I let end device connect and keep interacting, everything will be fine. If I disconnect (power off, reset,...) my end device and reconnect several times, my gateway (Pi + SX1301) sometimes cannot receive any packet, and I have to restart the gateway to make it work.

2. About the distance between gateway and devices:
I tried to test the distance, and my network only can work within few meters. Of course I know that I should upgrade the antenna, but can you tell me are there any parameters or method to improve in the code (such as TX power)?

Regard,
Thang

CanhThang, 2017-08-02:
Mr @mcoracin, can you help me?

closed 2021-06-24 by rbaldwin13