Lora-net/lora_gateway #57: 1301 Low-Power mode

kungccm, 2017-06-23:
Hi Sir,
There is a description of Sx1301 Low-Power mode in datasheet.
Do I need to config 1301 to low-power mode?
Or the FPGA has configured it to low-power mode already ?
Please advise.
Thanks.

mcoracin, 2017-06-28:
Hello,
There is no Low Power mode on the SX1301.
You can stick to what the lora_gateway hal project here does to properly initialize the SX1301.
Best regards

closed 2017-06-28 by mcoracin

kungccm, 2017-06-29:
Hi Sir,
In 1301 datasheet, it mentions low-power mode in "3.2.2 Setting the circuit is low-power mode".
http://www.semtech.com/images/datasheet/sx1301.pdf
Did I misunderstand about low-power mode?
Please advise. 
Thanks.

mcoracin, 2017-06-29:
Hello,
Yes, but the low-power wording in the datasheet is misleading. In fact this sequence is rather for booting-up the circuit. It is called low-power because the correlators are not started yet, so power consumption is lower. But once everything is started, we never go back in this state.
FOr booting up the SX1301, only the reset sequence is required, the BIST sequence is for production, don't bother about it.
Best regards

kungccm, 2017-06-29:
Hi Sir,

Understood. Thanks for your detailed reply.