Lora-net/lora_gateway #58: usage

alexanderplanar, 2017-06-28:
Hi everyone!
Excuse me for my, a little bit stupid and, maybe, off-topic question...
I'm trying to understand in general, how to use the LoRa-Gateway project.
I'm already made a LoRa-node device PCB and adapted the related project (LoRaMac-node) to my MCU, and there is no problems with it.
Now I use Microchip LoRa Gateway board to link my LoRa node with the LoRa environment (like LoRa Gateway bridge, network server and application server), and there no problems with it too.
The Microchip's gateway board is a bit buggy and has some restrictions, and so I need to made my own gateway instead of Microchip's.
Is there the full gateway PCB? As far as I understood, the LoRa-Gateway projects is only purposed to serve the HAL of SX130x, and must be controlled by gateway controller. Is'nt it? What the devices should be used for this purpose? Or, maybe, it is embedded linux computer with SPI interface to control the HAL? Please, explain me these points.

mcoracin, 2017-06-28:
Hello,
A typical LoRa Gateway consists in connecting a LoRa concentrator board (with SX1301s and radios) with a linux host board, through SPI interface.
The lora_gateway project, is the LoRa concentrator board driver that runs on the linux host board. It can be a Raspberry Pi, or anything else.

alexanderplanar, 2017-06-28:
**mcoracin**, thank you very much for fast reply!
So, as far as I understand, in my case I need:
 - linux-like board, aka. e.g. Raspberry PI;
 - LoRa Concentrator board, based on SX1301;
 - "[lora_gateway](https://github.com/Lora-net/lora_gateway)" project to link it with LoRa Concentrator board;
 - "[lora_packet_forwarder](https://github.com/Lora-net/packet_forwarder)" to link the lora_gateway project with the upper layer using Semtech UDP protocol, aka e.g. [LoRa Gateway Bridge](https://github.com/brocaar/lora-gateway-bridge)

Where I can find such LoRa Concentrator board? Can you give an example or link?
I try to evaluate what components and resources I need, and  I'm affraid do not understand fully what this board based on. I read that the board uses FPGA (field-programmable gate array), but the solutions I found in Internet does not have FPGA, e.g. Microchip LoRa Gateway has only SX1301 and a couple of SX1257. Maybe, It is required to order such board from Semtech?

Thanks in advance!

mcoracin, 2017-06-28:
Yes, you need all this, except the LoRa Gateway Bridge which is optional for now.

There are 2 versions of concentrator reference design, one with an FPGA, the other one without the FPGA, with lower capabilities (No Listen-Before-Talk, max TX power 20dBm in Europe).
The lora_gateway project supports both versions, with or without FPGA.

The iC880A board should work, I've never used it, but I think it is widely used in the TTN community.

In order to get access to Semtech's reference designs, you need to get in touch with your local Semtech sales contact:
http://www.semtech.com/contact/index.html#support

best regards

alexanderplanar, 2017-06-29:
Thank you for reply, **mcoracin**!
I'm newbie in this trend, I have been searching for some information about these boards ( with FPGA and without the one), but, unfortunately, with no success. Where I can read for some details, key features about there reference boards, or datasheets, before ordering it from Semtech? Could you give some links, if you are aware of?

_Ashamed to admit, I do not understand what the FPGA is used on the board for, due to I did not even see either the board without FPGA nor the board with the one. I can assume that FPGA is used to increase the performance of the board._

As for IC880A: Ok, thanks, I will try to find out some information from TTN community

Thanks!

mcoracin, 2017-06-29:
If you want to get more information about Semtech's reference board, you need to contact Semtech's sales, according to the region you reside in. Those boards are not publicly available.
Otherwise you can go with a public board, as the IMST one. You have a tutorial here:
https://github.com/ttn-zh/ic880a-gateway/wiki

The FPGA brings the following features:
- 27dBm TX power for Europe
- Listen-Before-Talk
- Background spectral scan

Best regards

alexanderplanar, 2017-06-29:
mcoracin, thank you, ok, I understood. I will get in touch with Semtech. 
Also, thank you for the link you gave. Now I have a couple options to choose: IMST and Semtech.

closed 2017-07-04 by mcoracin

alexanderplanar, 2017-09-21:
Hi again! I have extra question about this project, could you help me?

I've got BeagleBone Black board for this project, and I'm about of installing the Linux into the board.
As I understand, I will need to cross-compille the project for the board, or to build the project inside the board directly. Is there some recomendations of choosing the Linux kernel and version OS?

Thanks in advance!

mcoracin, 2017-09-21:
Hi,

There is no specific constraint on the kernel/distribution version. Debian or Ubuntu would be ok at least.

alexanderplanar, 2017-09-21:
Thanks! So I will try the default debian image for BeagleBone.

alexanderplanar, 2017-10-16:
Hi guys, I had installed Debian 8 on my BeagleBone board, then cloned the project "lora_gateway" and "packet_forwarder" and built these projects just using "make".
The projects do work, but I faced with little problem: the "stats" packets are being recognizing by LoraServer as invalid. The LoraServer throws a following warning message: 
> handle stats packet error: aggregate gateway stats error: insert or update aggregate error: pq: syntax error at or near ''on'' .

Can anyone to comment this error?

Just for information, I use brocaar's implementation of LoRa Server, maybe, it is LoRa  Server's problem, I do rule out this.

Stat. packet dump:

`{"stat":{"time":"2017-10-16 09:21:29 GMT","rxnb":1,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0}}`

```
0000   02 d0 1a 00 aa 55 5a 00 00 00 01 01 7b 22 73 74
0010   61 74 22 3a 7b 22 74 69 6d 65 22 3a 22 32 30 31
0020   37 2d 31 30 2d 31 36 20 30 39 3a 32 31 3a 32 39
0030   20 47 4d 54 22 2c 22 72 78 6e 62 22 3a 31 2c 22
0040   72 78 6f 6b 22 3a 30 2c 22 72 78 66 77 22 3a 30
0050   2c 22 61 63 6b 72 22 3a 30 2e 30 2c 22 64 77 6e
0060   62 22 3a 30 2c 22 74 78 6e 62 22 3a 30 7d 7d
```

Heltec-Aaron-Lee, 2018-08-23:
@alexanderplanar Hi Alexander, I have some problem while use BeagleBone to drive a SX1301 gateway, can you tell some more details about your project? 
e.g.
- The pin connection via SPI to gateway;
- Do I need some special config for Linux ? I use the newest Debian 9.4 2018-06-17 4GB SD IoT.
- Do I need fix the in lora_gateway project?

alexanderplanar, 2018-09-06:
Heltec-Aaron-Lee,
Hi, sorry for long waiting.
I use BeagleBoneBlack and LoRa Microchip Gateway Radio Board, so, your hardware config may be a little bit differ...
No, you do not need special config for **Linux**, but you need config for hardware overlay.
This board has pretty complicated configuration of hardware... 
I tried to enable SPI, read a lot of datasheets, spent a lot of time to understand why it does not work... As I realized, there no possibility to enable SPI0 and HDMI simultaneously. I had to disable HDMI in configuration to get possibility to enable SPI... But actually hdmi is still working... And the SPI1 is working instead of SPI0. Pfff!? As far as I understand, on my board SPI0 and SPI1 are mixed up. I found out it using oscilloscope..
I attached my config file: uEnv.txt. Place it into your sd-card images: **/boot/uEnv.txt**
[uEnv.txt](https://github.com/Lora-net/lora_gateway/files/2356856/uEnv.txt)

This file configures the hardware overlays in a boot time.
Anyway, you should read carefully the beaglebone datasheet for configuring your hardware.
So, due to this fact, you may need to change the SPI-name in the lora-gateway library:
macro `SPI_DEV_PATH`: I changed it to `/dev/spidev1.0` to select SPI-1 and ChipSelect=0.
This macro is located: _lora_gateway/libloragw/src/loragw_spi.native.c_

So, I used the official datasheet for RadioBoard: LoRa® Technology Gateway User’s Guide DS40001827A from Microchip, Table 2-1.
Each hardware pin in Beaglebone has a apropriate individual kernel-pin allowing to control the hardware (_beaglebone.kernel.pin_)
To list all the kernel pins use:
`sudo cat /sys/kernel/debug/pinctrl/44e10800.pinmux/pingroups`

I connected all lines in accordance with the document a refered above:
```
**spi.mosi**:
board.pin: 2
beaglebone.pin: P9_18, I2C1_SDA
beaglebone.kernel.pin: 86
**spi.miso**:
board.pin: 24
beaglebone.pin: P9_21, UART2_TXD
beaglebone.kernel.pin: 85
**spi.sck**:
board.pin: 26
beaglebone.pin: P9_22, UART2_RXD
beaglebone.kernel.pin: 84
**spi.~cs**:
board.pin: 28
beaglebone.pin: P9_17, I2C1_SCL
beaglebone.kernel.pin: 87
**gnd**:
board.pin: 6
beaglebone.pin: P9_46, GND
beaglebone.kernel.pin: -
**reset**:
board.pin: 18
beaglebone.pin: P9_42, GPIO_7
beaglebone.kernel.pin: 89
```
Next, I made a special shielded cable to connect the Beaglebone header to the Radioboard.
To validate the connection pins I used this:
[SPI_Ports](http://elinux.org/Beagleboard:Cape_Expansion_Headers#2_SPI_Ports)

I attached my **reset_lgw.sh** script to reset the SX1301 by linux-software.
[reset_lgw.txt](https://github.com/Lora-net/lora_gateway/files/2356858/reset_lgw.txt)

Heltec-Aaron-Lee, 2018-09-06:
Alexander

I had made it working... thanks for your reply.

alexanderplanar, 2018-09-06:
Okay, my congratulations :)
Did you come across any troubles with the hardware overlays?

Heltec-Aaron-Lee, 2018-09-06:
@alexanderplanar 
No... currently seems no trouble.