Lora-net/lora_gateway #59: send a packet to end-node from gateway directly

522623905, 2017-07-10:
Hello!I would like to know how to send a packet directly from the gateway to the end-node,.How can I do?

mcoracin, 2017-07-11:
WHat kind of end-node do you have? Is it a LoRaWAN class-A node with synchronized receive window, or a class-C node which can receive at any time?

522623905, 2017-07-14:
Class C mode. I use your \lora_gateway\util_tx_test\util_tx_test ,but nothing,as your see this data out:

522623905, 2017-07-14:
+REVD:RXWIN2,RSSI -66,SNR 34,PORT 0,RXSIZE 0,""
+REVD:DONE
+REVD:RXWIN2,RSSI -66,SNR 32,PORT 0,RXSIZE 0,""
+REVD:DONE
+REVD:RXWIN2,RSSI -67,SNR 31,PORT 0,RXSIZE 0,""
+REVD:DONE

522623905, 2017-07-14:
I just try to send a string,(eg . "test..."),but end-node receive " "

mcoracin, 2017-07-17:
Can you put here the exact command line you are using for util_tx_test?
Have you set the -i parameter to invert polarity? It is necessary for the end node to receive a packet sent from the gateway.

522623905, 2017-07-17:
yes,I have already setted -i.    ./util_tx_test -f 471.3 -s 9  -r 1255 -i  .     (And default bw=125,preamble length = 8 ,  payload length = 16  cr=4/5 ..)

mcoracin, 2017-07-17:
Does the gateway receive the packets sent by the end-node?
How is the private/public LoRa syncword configured on both the end-node and the gateway?

522623905, 2017-07-17:
private 0x12, public 0x34.

522623905, 2017-07-17:
If I send data directly from the server, through the gateway, the end-node can receive information. Now I would like to test directly from the gateway to send data to the end-node,but failed just like above the information

mcoracin, 2017-07-17:
util_tx_test does not send a LoRaWAN packet, it is just a LoRa packet with a raw payload, no LoRaWAN header.
I don't know which code is running on your end-node, so not sure about what the logs mean, but it seems that it actually receives packets, but it cannot get the payload, right?
Potentially because it checks for the payload MIC to be valid, but it is not, so it drops the payload.

util_tx_test is not a LoRaWAN packet tester.

522623905, 2017-07-17:
MIC,it's possible! But Is the MIC handled by the hardware?And I find  that the payload(txpkt.payload) is different from the previous if I send a string(eg,"AA") from server in lgw_send(txpkt).

mcoracin, 2017-07-17:
No, the MIC is software, purely LoRaWAN related. Please check the LoRaWAN specification to see how the payload should be constructed.

What you could do for testing also, is to grab the payload sent your server, and put it in the buffer sent by util_tx_test. So that you are sure that it is LoRaWAN compliant (expect the FCnt, hopefully the end-node SW will not reject the downlink if FCnt does not increase).

522623905, 2017-07-17:
And I compared the txpkt structure of the data, in addition to payload, the other contents are same. 
But if the payload I arbitrarily set some characters send out ,the end-node did not receive the payload. What is the reason for it to lose the payload?
Any suggestion will be appreciated!

mcoracin, 2017-09-21:
Is this issue still valid?

lcer93, 2018-06-19:
Yes, I have a similar problem. I receive e packet from a end-node and I'd like to reply without passing through a server, by artificially filling a txPkt and sending it to the end-node. But the node doesn't receive the reply.

sjain315, 2019-01-03:
@lcer93 and @522623905 
were you guys get any solution for this ?
I also have exact same issue

reissjason, 2019-01-03:
The network server is needed to author a correct response to the end-device.
A valid MIC can only be generated using the session keys.
The gateways are stateless and retain no information of the end-devices that are connected through it.

closed 2021-07-08 by rbaldwin13