Lora-net/lora_gateway #62: CALIBRATION FAILURE (STATUS = 0)

wushichuan, 2017-07-20:
Hi , I enconter some error when I setup the **util_tx_test**，**./util_tx_test -r 1257 -f 886.5**:

**> ERROR: CALIBRATION FAILURE (STATUS = 0)
> WARNING: calibration could not access radio A
> WARNING: calibration could not access radio B
> WARNING: problem in calibration of radio A for image rejection
> WARNING: problem in calibration of radio B for image rejection
> WARNING: problem in calibration of radio A for TX imbalance**

how can I slove it ?

mcoracin, 2017-07-20:
Hi, do you get those errors every time you run the calibration sequence?

wushichuan, 2017-07-21:
@mcoracin  yes, I get those errors every time,

wushichuan, 2017-07-21:
HI, @mcoracin , my host conctrol is mtk7621,

wushichuan, 2017-07-21:
@mcoracin  I wonder that the calibration sequence is necessary? can I ignore this step ?

mcoracin, 2017-07-21:
@wushichuan yes, the calibration sequence is necessary to have good radio performances.

Difficult for me to say what is the problem without having more info about your setup.
Which gateway board (SX1301) are you using?
How is it connected to the host?
Which version of the gateway software are you running?

wushichuan, 2017-07-21:
@mcoracin 
1.The gateway board called RHFOM301, which contains SX1301 chip and tow 1257 chips.
2. The spi interface,  MOSI, MISO, CLOCK, SELECT, connected ok, the power supply is ok,
3.The gateway version is  v4.0.1.

I had write a blog about the hardware detail. more detials click here:  
[http://blog.csdn.net/wsclinux/article/details/75665008](url)

thank you for your help.

mcoracin, 2017-07-21:
@wushichuan , have you run the util_spi_stress program to check that your SPI connection is reliable?

./util_spi_stress -t 4

It should run for hours with no error.

mcoracin, 2017-07-21:
@wushichuan also, do you control the RESET pin of the RHFOM301 from your mtk7621?

wushichuan, 2017-07-22:
@mcoracin 
./util_spi_stress -t 4 
output:
**BURST WRITE: to trans 1024 # chunk 1024 # transferred 1024 
BURST WRITE: to trans 2 # chunk 2 # transferred 2 
did a 1024-byte R/W on a data buffer with no error**
the 1024 bytes cut to 2 times send due to my diriver, but the busrt wr and busrt rd checking is ok.

In addition, I had no connected the reset pin,  it is neccesary?  Maybe it causes the err. I will verify it right now!

mcoracin, 2017-07-24:
@wushichuan yes, the reset pin is necessary. If it is not connected, it could explain your issue!

wushichuan, 2017-07-24:
@mcoracin I had connected reset pin，and reset it according to the power-up sequence，but it takes no effort，

wushichuan, 2017-07-24:
hi @mcoracin ，here is some code from file loragw_reg.c (line 142-143):
```
    {-1,126,0,0,8,1,1},       /* CHIP_ID */
    {-1,127,0,0,1,0,1},       /* EMERGENCY_FORCE_HOST_CTRL */
```
the 1301 datasheet indicates the "CHIP_ID" register address is 0x7F (127), but the code indicates the 0x7F (127) address is “ EMERGENCY_FORCE_HOST_CTRL”，they are in conflict. Is it a mistake？

wushichuan, 2017-07-27:
hi @mcoracin , I don't konw why some registers can't be found in the sx1301 datasheet, such as  
 the LGW_MCU_PROM_ADDR,  and the code ` **lgw_reg_w(LGW_MCU_PROM_ADDR, 0);**` can not be execute sucessfully, Maybe it causes the errors.

wushichuan, 2017-08-11:
the issue had been sloved, it caused by the driver

closed 2017-08-11 by wushichuan

zyzyzyzzyy, 2017-09-06:
@wushichuan 
I have the same problem as you, can you tell me what caused STATUS = 0? How did you solve the problem?

kitplummer, 2020-05-21:
I get to about here regularly to cause the CALIBRATION ERROR:

```
Cycle 275 > did a 1024-byte R/W on a data buffer with no error
Cycle 276 > error during the buffer comparison
Written values:
 EB  E5  A5  A7  20  20  3F  5F  7B  6A  CD  C8  DD  D7  7B  86
```