Lora-net/lora_gateway #65: Packet drop issue with multiple devices

vikasrohal, 2017-08-03:
Hi,
I am testing with 5 End devices and observe few packets are drop but End devices Send these packets 8 times and no single packet received by gateway 

As per attached log FCNT=7 of (0000009) device address droped but End Device Send this packet 8 times  with CRC ok (as per util_pkt_logger log).

Its happen with other devices also.

Lora Gateway version - v5.0.1
Lora packet forwarder - v4.0.1
Lora Mac Master - v4.3.2

[2_Aug_5_devices_2nd_test.txt](https://github.com/Lora-net/lora_gateway/files/1196059/2_Aug_5_devices_2nd_test.txt)

xjh200858, 2017-08-16:
Yes,I have the same problem.
src/jitqueue.c:280:jit_enqueue(): ERROR: Packet (type=0) REJECTED, collision with packet already programmed at 1551992540 (1552155148)

satideur, 2017-09-14:
I think it would be helpful if you attach the Tx log because the packets could be colliding

closed 2021-06-24 by rbaldwin13