Lora-net/lora_gateway #66: LoRa Gateway setup failed with ic880A-spi and RPi 3 Model B

rlokanathan, 2017-08-04:
I am using the SX1301 chip for the ic880A-spi model LoRa module with my Raspberry Pi 3 Model B. I am trying to setup my Raspberry Pi as my gateway. I have followed the instruction from the github and also from the iC880A-spi quick guide pdf document. But still its failing.

I am getting error messages like `lgw_connect:555: ERROR: NOT EXPECTED CHIP VERSION (v0)` and also `ERROR: CONCENTRATOR UNCONNECTED` 

I am getting this error when i run the **test_loragw_reg** from the **libloragw**.

I have read alot of blogs and forums but neither helped me to solve this issue. I have checked my RPi3 wiring with the ic880a-spi. All the wires setup looks good. Even the power supply also I checked its good.

Are there anyways to solve this issue. I am new to this LoRa technology.

szhansen, 2017-08-17:
hi,where do you get  ic880A board ? how much ?
I want to buy one too.

dmamalis, 2017-08-18:
@rlokanathan unfortunately this has been around for quite a long time and indeed there is no real answer. I have had the same problems before and the only solution (which I actually consider a workaround rather than a solution) is referred here #26 

However, even that doesn't always work! If you check around for rpi3 SPI issues you will see that this is probalby an RPI3 specific problem rather than a LoRa Hardware one. Check the rpi github issues.

mcoracin, 2017-09-20:
It is a SPI issue. Check the SPI configuration on the RPi.

closed 2017-09-20 by mcoracin

LoRa-YvAd, 2017-09-22:
https://github.com/jpmeijers/ttn-resin-gateway-rpi worked for me