Lora-net/lora_gateway #67: lgw_connect:532: INFO: no FPGA detected or version not supported (v0) - version 5.0.1 --Any help

narioinc, 2017-08-05:
Hi Guys i am using the library version 5.0.1 for the lora gateway with the RAK wireless RAK831 lora gateway module. I am using a Linkit 7688 module with the SPI half duplex fix firmware (also only supports 16 byte chunks of payload). I have double checked the pin connection and they seem ok (GND connections, voltage level, and SPI connection are ok )

What i am seeing is this error when i try to do a test_loragw_reg:
./test_loragw_reg 
Beginning of test for loragw_reg.c
Note: SPI port opened and configured ok
Note: SPI read success
lgw_connect:532: INFO: no FPGA detected or version not supported (v0)
Note: SPI read success
lgw_connect:555: ERROR: NOT EXPECTED CHIP VERSION (v0)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
IMPLICIT_PAYLOAD_LENGHT = 2003477408 (should be 197)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
FRAME_SYNCH_PEAK2_POS = 2003477408 (should be 11)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
PREAMBLE_SYMB1_NB = 2003477408 (should be 49253)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
ADJUST_MODEM_START_OFFSET_SF12_RDX4 = 2003477408 (should be 3173)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
IF_FREQ_1 = 2003477408 (should be -1947)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
Note: SPI port closed
Note: success disconnecting the concentrator
End of test for loragw_reg.c

Am i using the wrong version of the library or do i need to change something explicitly in the c/.h files to avoid this error?

Any help would be appreciated. 

an test_loragw_spi gives this output:
root@mylinkit:~/lora# ./test_loragw_spi 
Beginning of test for loragw_spi.c
Note: SPI port opened and configured ok
WARNING: SPI address > 127
Note: SPI write success
Note: SPI read success
lgw_spi_wb:309: BURST WRITE: to trans 16 # chunk 16 # transferred 16 
Note: SPI burst write success
lgw_spi_rb:371: BURST READ: to trans 16 # chunk 16 # transferred 16 
Note: SPI burst read success
lgw_spi_wb:309: BURST WRITE: to trans 2500 # chunk 1024 # transferred -2 
lgw_spi_wb:309: BURST WRITE: to trans 1476 # chunk 1024 # transferred -4 
lgw_spi_wb:309: BURST WRITE: to trans 452 # chunk 452 # transferred -6 
ERROR: SPI BURST WRITE FAILURE
lgw_spi_rb:371: BURST READ: to trans 2500 # chunk 1024 # transferred -2 
lgw_spi_rb:371: BURST READ: to trans 1476 # chunk 1024 # transferred -4 
lgw_spi_rb:371: BURST READ: to trans 452 # chunk 452 # transferred -6 
ERROR: SPI BURST READ FAILURE
Note: SPI read success
data received (simple read): 0
Note: SPI port closed
End of test for loragw_spi.c

which is expected given the limitation of the mt7688 chipset (16 bytes paylond chunk only) and the silicon level bug in SPI

matchxiot, 2017-08-29:
Your PCB has no FPGA, so it is normal

closed 2017-09-20 by mcoracin

matchxiot, 2020-07-07:
Hi Michael,

Hope everything is fine on your side!
I have a quick question.

with regard to the duty cycle in EU, US or Korea, there is no calculation to comply with it in packet forwarder right?

And we didn't see it is calculated in Network Server either.

Any hint?

Xin Hu (Sheen)

CEO

MatchX GmbH

Brückenstraße 4

10179 Berlin

T: +49 (0)3054908830 | M: +49 (0)1638924655

http://www.matchx.ai/ | mailto:sheen@matchx.ai

Registered in : Berlin, Registergericht: Amtsgericht Charlottenburg.

Registration Number: HRB 182407; Geschäftsführer: Xin Hu

---- On Wed, 20 Sep 2017 16:57:06 +0200 Michael Coracin <notifications@github.com> wrote ----

Closed https://github.com/Lora-net/lora_gateway/issues/67.