Lora-net/lora_gateway #68: Where can i get the Schematic diagram of SX125x+SX1301 demo board ?

szhansen, 2017-08-12:
anybody could tell me ?

how can i get the demo board  or the  Schematic diagram ?

Thank you so much...

nestorayuso, 2017-08-13:
the reference design is under NDA from semtech, but you cand find schematic at http://ww1.microchip.com/downloads/en/DeviceDoc/40001827A.pdf
or you can look at the pico gateway reference design, it is almost the same adding a STM32 MCU
http://www.semtech.com/apps/filedown/down.php?file=sx1308p868gw_picocell_gtw_e381v02b_eu_folder.zip

szhansen, 2017-08-13:
@nestorayuso 
thank you so much.
I noticed that the pico gateway use SX1308 not SX1301, there is some question about this:
1. what's the difference between SX1308 and SX1301 ?
2. what's the function of STM32 ? I think that in SX1301 GW, we can access  SX1301 by SPI bus directly from such as raspberry pi ?  but how about SX1308 ? Raspberyy pi SPI ---> STM32 SPI ---> SX1308 SPI ?

Thanks again for  your help ,I am the fresh ...

nestorayuso, 2017-08-16:
1. both are basically the same, only changes in operational temperature range and power requirements.

2. in gateway reference design, both software modules lora-gateway and packet_forwarder run on linux platform (raspberry-pi for example), and a low-latency/high speed SPI is needed (USB was avoided).
in pico-gateway reference design, the lora-gateway software module runs on a STM32 and packet_forwarder run on linux platform (raspberry-pi for example), in this case,  low-latency/high speed SPI is used between SX130x and STM32, and UART/USB CDC between STM32 and raspberry-pi because latency is not critical now.

szhansen, 2017-08-16:
@nestorayuso thank you so much~~~
It seems that  pico-gw  have better performance than SX1301.

szhansen, 2017-08-17:
@nestorayuso 
the packet_forwarder need server,
if i just install the server on my raspberry-pi for test, which server software can i use  ?

thank you.

szhansen, 2017-08-17:
@nestorayuso 
I have review the picoGW-mcu code, and find the mcu can use UART or USB to connect with RPi:
`#ifdef  USE_UART 
CMDMANAGER::CMDMANAGER(PinName Tx, PinName Rx) 
{
    kill = false;
    ActiveInterface = (INTERFACE *) new COMUART (Tx, Rx);
    ActiveCom = ISUARTINTERFACE;
}
#else
CMDMANAGER::CMDMANAGER(PinName Tx, PinName Rx)
{
    kill = false;
    ActiveInterface = (INTERFACE *) new COMUSB ();
    ActiveCom = ISUSBINTERFACE;
}
#endif`

But in the code of picoGW_packet_forwarder, only UART (/dev/ttyACM0)  can be used,no USB,
Why ???

mcoracin, 2017-09-20:
If the picoGW_mcu is compiled in USB mode, when inserted on the USB port it will be mounted on /dev/ttyACM0 on the RPi. Even though it is USB, it is a serial device.

Anyway, you can indicate to the packet forwarder which device to be used with -d option.
Use dmesg to get the device asssociated to the picoGW.

closed 2017-09-20 by mcoracin

Serjaru, 2018-10-05:
> the reference design is under NDA from semtech, but you cand find schematic at http://ww1.microchip.com/downloads/en/DeviceDoc/40001827A.pdf
> or you can look at the pico gateway reference design, it [...]

Link broken. Anybody, can send me?

nestorayuso, 2018-10-05:
https://www.semtech.com/products/wireless-rf/lora-gateways/SX1308