Lora-net/lora_gateway #69: Receive packets with implicit header reception

0x4C4A, 2017-08-22:
I wasn't able to find any way to enable implicit header mode. Is there any way to receive such messages?
Since implicit header mode gives some nice time-on-air (and therefore energy-efficiency) benefits, we expected it to be available here, just like on the SX1272's.

mcoracin, 2017-08-22:
Only the SingleSF LoRa modem can be used in implicit header mode, so you can have only one channel in that mode.
If you want to do that, you can use the following registers:
```
lgw_reg_w(LGW_MBWSSF_RATE_SF,9);
lgw_reg_w(LGW_MBWSSF_IMPLICIT_HEADER,1); /* no header */
lgw_reg_w(LGW_MBWSSF_IMPLICIT_CRC_EN,0);
lgw_reg_w(LGW_MBWSSF_IMPLICIT_CODING_RATE,1);
lgw_reg_w(LGW_MBWSSF_IMPLICIT_PAYLOAD_LENGHT,17);
```

0x4C4A, 2017-08-22:
So the following registers from `loragw_reg.h` are some sort of a deprecated thing?
```
LGW_IMPLICIT_CRC_EN 
LGW_IMPLICIT_CODING_RATE 
LGW_IMPLICIT_PAYLOAD_LENGHT
```

0x4C4A, 2017-08-23:
I see, thanks a lot for the quick responses!

closed 2017-08-23 by 0x4C4A