Lora-net/lora_gateway #70: How can I get the time of packet arrival in Gateway from End-Node by NanoSeconds?

moonhyunkoo6356, 2017-08-23:
Hello,

I'm analyzing the packet receive function(lgw_receive) in libloragw library to get the packet arrival time by nanoseconds from end-node to gateway. But I found that it’s getting the microsecond time.
(p->count_us = raw_timestamp – timestamp_correction)
The ‘p->count_us’ is the packet arrival time and it's calculated by microseconds.

I really need nanoseconds time of packet arrival to do my projects.
should I modify library code? or change some options?
please help me

Thank you

mcoracin, 2017-08-24:
Hello,

The packet timestamp comes from a SX1301 counter which has a microsecond resolution, so you can't have anything more precise than that. It is not done by the HAL itself, it is hardware.

moonhyunkoo6356, 2017-09-11:
Is there any idea using GSP Module?

mcoracin, 2017-09-20:
Not possible with this gateway.

closed 2017-09-20 by mcoracin