Lora-net/lora_gateway #74: error message for fw 1

allionWilliam, 2017-09-12:
Hi everyone

I have run my SX1301 gateway util_tx_test and meet a error message down below.
What is the exactly the fw 1 it is?

lora@lora-iot-sk /opt/lora-net/gateway/lora_gateway/util_tx_test $ ./util_tx_test  -f 868.5 -r 1257

Sending -1 packets on 868500000 Hz (BW 125 kHz, SF 10, CR 1, 16 bytes payload, 8 symbols preamble) at 14 dBm, with 1000 ms between each

ERROR: Failed to load fw 1
ERROR: Version of calibration firmware not expected, actual:0 expected:2
ERROR: failed to start the concentrator

mcoracin, 2017-09-12:
Hi,

Have you set the SX1301 out of reset through the reset pin?

allionWilliam, 2017-09-13:
Hi mcoracin

Where can I reset the reset pin?
Use program or other ways?
Thank for your reply.

mcoracin, 2017-09-13:
Hi,

Which sx1301 board do you have? Which host CPU? how both are connected?

allionWilliam, 2017-09-13:
Hi mcoracin

It is a sx1301 start kit.
And use the RASPBERRY PI 2 MODEL B to control whole system.
The connect way is just like the attachment.
[Lora.docx](https://github.com/Lora-net/lora_gateway/files/1298682/Lora.docx)

Many thanks

mcoracin, 2017-09-13:
ok, so if it is a starter kit, you should be able to use this script to set the sx1301 out of reset:
https://github.com/Lora-net/lora_gateway/blob/master/reset_lgw.sh

This should be run before launching the packet forwarder or any other tool.

allionWilliam, 2017-09-13:
Cheers!!!
Impressive, it really work.
But one more question, it is normal to show " CRC error" in the INFO dialogue?

INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1
INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "localhost"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
-->INFO: packets received with a CRC error will NOT be forwarded
-->INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: [main] concentrator started, packet can now be received

mcoracin, 2017-09-13:
Yes, it is intended to work! ;)

The CRC error line that you mention is just an information that the packet forwarder is configured NOT to forward packets received with a CRC error. You can choose to forward it by modifying the global_conf.json.

closed 2017-09-13 by allionWilliam

rego21, 2017-12-27:
Hello,

I have the same error and also ran the reset script, but is showing the same message... Im using a Raspberry Pi 2 with the Microchip Concentrator from their Development Kit. Any tips?

SaiPrakash14, 2018-01-04:
I am also having the same issue Please help me how to solve this

arihantdaga, 2018-03-16:
I am facing the same issue... 
Nothing is working.. 
Can somebody help.. ?

arihantdaga, 2018-03-16:
I solved My issue. 
I went through this video and did exactly the same thing. Once done with that everything is working fine.
Posting here, It might help others. 
https://www.youtube.com/watch?time_continue=295&v=gYJTDrRstZE