Lora-net/lora_gateway #77: Periodic Pauses under High Traffic Load

satideur, 2017-09-14:
Under a stress test (high traffic: simultaneus transmissions from several devices (>6) on different SFs and Freqs) of the Packet Forwarder/Logger I have found **periodic pauses of several seconds** in the demodulation of packets. During these pauses, **no packet at all** is logged / forwarded. I have already checked that the process still runs. After several seconds (sometimes one second, sometimes 10 and more seconds), depending on how high the traffic load was, packets are logged/forwarded again  normally until the next pause comes.

As I understand from Semtechs 1301's datasheet, one could be able to receive a constant stream of 8 demodulated packets:

> The SX1301 can detect simultaneously preambles corresponding to all data rates on all IF0 to IF7
channels. However it cannot demodulate more than 8 packets simultaneously. This is because the
SX1301 architecture separates the preamble detection and acquisition task from the demodulation
process. The number of simultaneous demodulation (in this case 8) is an arbitrary system parameter
and may be set to any value for a customer specific circuit.

Unfortunately until now my tests show that packets from only aprox. **4 RF chains in average** are buffered (8 chains used intermitently). In the Packet [Log.txt](https://github.com/Lora-net/lora_gateway/files/1302452/Log.txt) one can see some long pauses 11:04:53-11:05:09, 11:05:10-11:05:17, 11:05:18-11:06:02. Following only an extract of the pauses:

gateway ID,"node MAC","UTC timestamp","us count","frequency","RF chain","RX chain","status","size","modulation","bandwidth","datarate","coderate","RSSI","SNR","payload"
...
AAAAAAAAAAAAAAAA,"","2017-09-07 11:04:53.030Z", 909352972, 868300000,0, 1,"CRC_OK" , 22,"LORA",125000,"SF12"  ,"4/5",-51, +8.0,"00000000-10070061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:04:53.216Z", 909541092, 869525000,1, 6,"CRC_OK" , 22,"LORA",125000,"SF12"  ,"4/5",-69, +9.2,"00000000-10050061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:05:09.730Z", 926081219, 868100000,0, 0,"CRC_OK" , 22,"LORA",125000,"SF7"   ,"4/5",-43, +7.5,"00000000-10000061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:05:09.810Z", 926159892, 867700000,0, 3,"CRC_OK" , 22,"LORA",125000,"SF8"   ,"4/5",-47, +8.0,"00000000-10010061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
...
AAAAAAAAAAAAAAAA,"","2017-09-07 11:05:10.502Z", 926824076, 868300000,0, 1,"CRC_OK" , 22,"LORA",125000,"SF12"  ,"4/5",-48,+10.0,"00000000-10070061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:05:10.688Z", 927011076, 869525000,1, 6,"CRC_OK" , 22,"LORA",125000,"SF12"  ,"4/5",-70, +7.8,"00000000-10050061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:05:17.115Z", 933464964, 867700000,0, 3,"CRC_OK" , 22,"LORA",125000,"SF8"   ,"4/5",-48, +8.0,"00000000-10010061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:05:17.159Z", 933508571, 868100000,0, 0,"CRC_OK" , 22,"LORA",125000,"SF7"   ,"4/5",-43, +8.0,"00000000-10000061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
...
AAAAAAAAAAAAAAAA,"","2017-09-07 11:05:17.947Z", 934292588, 868500000,0, 2,"CRC_OK" , 22,"LORA",125000,"SF10"  ,"4/5",-48, +8.8,"00000000-10030061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:05:18.303Z", 934625220, 869525000,1, 6,"CRC_OK" , 22,"LORA",125000,"SF12"  ,"4/5",-67, +4.8,"00000000-10050061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:06:02.436Z", 978784763, 868100000,0, 0,"CRC_OK" , 22,"LORA",125000,"SF7"   ,"4/5",-45, +7.2,"00000000-10000061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
AAAAAAAAAAAAAAAA,"","2017-09-07 11:06:02.548Z", 978899099, 868100000,0, 0,"CRC_OK" , 22,"LORA",125000,"SF7"   ,"4/5",-45, +7.2,"00000000-10000061-AAAAAAAA-AAAAAAAA-AAAAAAAA-AAAA"
..

I would like to understand **why the pauses are happening** and if there is a maximum capacity of the chip in terms of Time On Air or Bytes per unit of time. Additionally, from Semtech's documentation one could change the maximum of simultaneus demodulated packets but it is not clear which register to modify and is its maximum value.

Thanks

mcoracin, 2017-09-20:
Hello,

The SX1301 is able to receive 8 multiSF channels in parallel (+ 1 monoSF + 1 FSK) with no interruption, as long as there is enough room left in the SX1301 Data buffer (1024 bytes).

How regularly is the pause happening?

I've run a test of more than 10 minutes, with 8 channels of the gateway set to the same frequency, and a end-point sending SF7 packets of 12-bytes length continuously.

I've sent 20000 packets, and received 159302, so a PER below 1% which is expected.
No pause happened during that time.

Do you need longer test to reproduce your issue?
Could you launch the same test, using the util_pkt_logger tool and see it you get same result that me?

mcoracin, 2017-09-20:
The corresponding CSV file, if of any interest.
[pktlog_AA555A0000000101_20170920T135844Z.zip](https://github.com/Lora-net/lora_gateway/files/1318029/pktlog_AA555A0000000101_20170920T135844Z.zip)

satideur, 2017-09-20:
Thank you very much.

Could you please try the same test but this time using 8 nodes, each one for a different frequency? In the test everything works ok but it differs from the scenario of the problem, in which there are several end devices transmitting simultaneusly. One should be able to see pauses in less than a minute when using 3 or more nodes transmitting constantly

mcoracin, 2017-09-21:
You confirm that it works ok on your side when using same test scenario than me?

Can you describe more precisely your test setup?
How often do your nodes transmit? Continuous SF7 with no pause between each?

mcoracin, 2017-09-21:
I've tested the following:
3 end-points, sending continuously SF7 packets of 22-bytes:
- 1 end-point on 866.1
- 1 end-pont on 866.3
- 1 end-pont on 866.5
Each end-points sends 2000 packets.

The gateway received 5866 packets instead of 6000, which is an expected PER of around 3%.
But I saw no pause...

I'll try with a fourth one.

Is it closer to what you are testing?

satideur, 2017-09-21:
Yes, this is what I tested and described in issue [#55](https://github.com/Lora-net/packet_forwarder/issues/55) from the packet forwarder.

I could reproduce your tests and found that everything works ok and have the same results as you.

When sending with only one node 22 bytes payload and all radios from the GW in the same frequency: I have the same results as you: sending every 60ms (57 ms time on air), I receive almost 100% of the packets (7184 Rx out of 899 Tx). See [22BytesPayload.txt](https://github.com/Lora-net/lora_gateway/files/1320847/22BytesPayload.txt)

When sending 255 bytes payload (period bet. Tx: 405ms, Time on Air: 400 ms), it works as expected, only delivering 3 messages because of the buffer size and the fact that all messages come simultaneusly: 426 Rx out of 143 Tx Msgs. Again almost 100% of the packets
[255BytesPayload.txt](https://github.com/Lora-net/lora_gateway/files/1320849/255BytesPayload.txt)

Please let me know how do the tests with the 4th node go. This is where problems start to happen

mcoracin, 2017-09-21:
Please find attached the test results with 4 end-points, sending 22-bytes payload at SF7 continuously.
866.1, 866.3, 866.5, 866.7

Received 7866 packets instead of 8000 (2% PER), no pause.

[pktlog_AA555A0000000101_20170921T120858Z.zip](https://github.com/Lora-net/lora_gateway/files/1321062/pktlog_AA555A0000000101_20170921T120858Z.zip)

mcoracin, 2017-09-21:
Which version of the software are you using, on which gateway?

satideur, 2017-09-21:
Can you use 255 bytes payload, this is what I tested, I will test also with 22 bytes payload to see if I get the same as you

GW: Kerlink, Packet Forwarder and utils based on in HAL v4,1,3 and v3.1.0

mcoracin, 2017-09-21:
Ok, I've reproduced your issue with 255 bytes payload. :-/
We're investigating and let you know the result.

Thank you

satideur, 2017-12-04:
Hello @mcoracin , do you have any new information regarding this issue? Thanks

dubanoze, 2017-12-18:
Hi @mcoracin, any new about this issue?
Thank you.

Dollar-Yao, 2018-09-19:
> Please find attached the test results with 4 end-points, sending 22-bytes payload at SF7 continuously.
> 866.1, 866.3, 866.5, 866.7
>
> Received 7866 packets instead of 8000 (2% PER), no pause. [...]

Hi, mcoracin, how about this issue? I'm also care about it. I checked your log, you use 4 nodes with same SF, and the RSSI is so strong, is there won't occur a collision? Thank you

satideur, 2019-02-18:
Hello @mcoracin, have you given up on this issue?

closed 2021-07-08 by rbaldwin13