Lora-net/lora_gateway #79: LoRa Semtech sx1301

ianmucil, 2017-09-28:
Hello there,
I'm student, i got development kit semtech SX1301 from my teacher.
But i don't how to start it, i didn't get the manual book or something like that.
Can anybody here help me to start it?like how to program or to communicate or anything to start.

thanks..

mcoracin, 2017-09-28:
Hello,

You mean you got a Gateway IoT Starter Kit, right?
Please find below the corresponding user guide:
[IOT868STK_UG_V1.0_STD.pdf](https://github.com/Lora-net/lora_gateway/files/1340814/IOT868STK_UG_V1.0_STD.pdf)

There are also some information that could be of interest here:
[IoT Start Kit user guide](https://github.com/Lora-net/packet_forwarder/wiki/Use-with-Raspberry-Pi)

ianmucil, 2017-10-04:
hai,
how can i connect my sx1276 to my gateway sx1301?
so that i can start simple demonstration as described in the user guide.

thx

mcoracin, 2017-11-09:
Sorry, but this question is a bit vague or too large to be answered here. There are plenty of tutorial on the web to learn to do that.

closed 2017-11-09 by mcoracin