Lora-net/lora_gateway #84: Question on SX1301

moonhyunkoo6356, 2017-10-22:
Hello.

I purchased LoRa IOT Starter Kits(SX1301 and Raspbbery Pi) last month and have a question about this.
 
We want to execute test example ‘test_loragw_gps’ and ‘gps_pkt_fwd’ in lora-net package. But I think GSP is not enabled because of something wrong in local_conf.json and global_conf.json files. The problem screen is below.

['test_lora_gps']
[image]

['gps_pkt_fwd']
[image]

WARNING: [gps] could not get UTC time from GPS

is our main issue and we really want to configure out this. 
for doing this, Should I change configure in my local_conf.json file? 
or another issue about hardware related with GSP?
My local_conf.json file is below
[image]

Please help us about this. We really need your help.

Thank you.

mcoracin, 2017-11-09:
What do you get if you type the following command?
> cat /dev/ttyAMA0

mcoracin, 2017-12-18:
no answer, closing the issue.

closed 2017-12-18 by mcoracin