Lora-net/lora_gateway #86: LoRa receiving channels

nkaramolegos, 2017-11-06:
As I had understood these channels 868.1, 868.3, 868.5 are the minimum set that all gateways should always be listening (for join etc.). Also, there is a channel (868.8) which supports only FSK modulation, used for high speed transfer. However, each network gateway is listening at the same time a particular set of 9 channels, so which are the frequencies of the 5 remaining channels that this gateway implementation receives?

larkolab, 2017-11-09:
A SX1301 based gateway has 10 channels which can be configured:
- 8 LoRa multi-SF channels
- 1 LoRa mono-SF channel
- 1 FSK channel

As you said, there are 3 default LoRa channels which must be set for joining.
The other channels can be freely attributed.

closed 2017-11-09 by mcoracin

nkaramolegos, 2017-11-11:
Thank you for your response. Thus, the other channels can be freely attributed, but where in the code or in the configuration files?

mcoracin, 2017-11-13:
It is defined in the global_conf.json file, in the chan_multiSF_x objects.
You need to change the 'if' values and you may have to change the 'radio_x/freq' field depending on the channel frequency you want.

In the default global_conf.json file you have:
        "radio_0": {
            ...
            "freq": 867500000,
            ...
        },
        "radio_1": {
            ...
            "freq": 868500000,
            ...
        },
        "chan_multiSF_0": {
            /* Lora MAC channel, 125kHz, all SF, 868.1 MHz */
            ...
            "radio": 1,
            "if": -400000
        },
        "chan_multiSF_1": {
            /* Lora MAC channel, 125kHz, all SF, 868.3 MHz */
            ...
            "radio": 1,
            "if": -200000
        },

This indicates that channel 0 frequency is set to the center freq of the radio 1 minus 200000Hz (868500000 - 200000 = 868.3MHz).

etc...

nkaramolegos, 2017-11-13:
You are the best. Thank you

nkaramolegos, 2017-12-03:
Hello, I have one more question. Is there the capability to set the BW of the chan_multiSF channels to something different than  125 KHz for example 250 or 500 KHz using the variable bandwidth?

mcoracin, 2017-12-04:
Hello,
No, it's not possible, the multiSF channels are 125KHz only.