Lora-net/lora_gateway #89: multiple gateways with single node

setty203, 2017-11-15:
hi, this is ashok.

i want to send data to multiple gateways from the single Node.

can anyone help me regarding that?

mcoracin, 2017-11-15:
Hi,

There is no association between the node and the gateway, when a node talks, all the gateways around will receive the data and forward to the server.

So there is nothing specific to do for that.

closed 2017-11-15 by mcoracin

setty203, 2017-11-15:
dear Mcoracin, 

thanks for the response, actually i am working on Triangulation method for identification of the Node Location using RSSI with 3 Gateways and multiple Nodes, but the Node is able to send the data to only one gateway instead of 3.

could you help me regarding that?

mcoracin, 2017-11-15:
Why is the node able to send the data to only one gateway?
Could you elaborate?

setty203, 2017-11-16:
dear Mcoracin,

**Why is the node able to send the data to only one gateway?
based on the distance it was sending the packet to the nearest gateway.**

- actually, I am working one the Geolocation application by using RSSI. 

- when my Node was sending the packets only one Gateway is receiving, but I want all the gateways must receive the packet.

mcoracin, 2017-11-16:
Which SF value is your node using to send the packets?

mcoracin, 2017-11-16:
So, if you already use SF12, and only one gateway receives your end-node data, there is not much to do I'm afraid...

Do you own the gateways?

I understand that you need your end-node to be received by 3 gateways for your localization, but for this, the end-node needs to be close enough to the 3 gateways... Is it the case?

setty203, 2017-11-17:
Dear mcoracin,

**So, if you already use SF12, and only one gateway receives your end-node data, there is not much to do I'm afraid...**
in this what is the preferable SF for the Single Channel gateway?

**Do you own the gateways?**
yes, 3 Single Channel Gateways using SX1272 with the RASPBERRY PI ZERO.

**I understand that you need your end-node to be received by 3 gateways for your localization, but for this, the end-node needs to be close enough to the 3 gateways... Is it the case?**

yes.

mcoracin, 2017-11-17:
There is no preferable SF for a single channel gateway, it just needs to be inline with the SF used by the end-node and consistent with the distance between the end-node and the gateway.

For a localization based on RSSI, I would not recommend to use SF12, the RSSI is less precise.

For you tests, put your 3 gateways within a reasonable distance of your end-node and configure everyone in SF7.

If all gateways are configured the same way, you should receive the packet sent by the end-node on all gateways.

setty203, 2017-11-17:
Dear mcoracin,

thanks for the update. let me check with that.