Lora-net/lora_gateway #91: LoRa Gateway for Class A device handling

GauravPatni, 2017-11-22:
Dear Sir,
We are implementing LoRa Gateway for Class A device using  iC880A concentrator.

we are referring following code from gateway :

https://github.com/Lora-net/lora_gateway
https://github.com/Lora-net/lora_gateway/tree/master/libloragw/src

Note that we are neither using  Raspberry Pi nor Linux . We are implementing this on our custom designed MCU based board.

We have doubt in following code section :

**doubt 1:** 

> What is teh significance of " lgw_get_tx_start_delay() " ? why it is used ? As we are using our own custom MCU based board then still we need to use this function ?

FILE NAME : "loragw_hal.c"

FNCTION NAME : "int lgw_send(struct lgw_pkt_tx_s pkt_data)"

```

/* Get the TX start delay to be applied for this TX */
    tx_start_delay = lgw_get_tx_start_delay(tx_notch_enable, pkt_data.bandwidth);
	

/* metadata 3 to 6, timestamp trigger value */
/* TX state machine must be triggered at (T0 - lgw_i_tx_start_delay_us) for packet to start being emitted at T0 */
    if (pkt_data.tx_mode == TIMESTAMPED)
    {
		count_trig = pkt_data.count_us + (uint32_t)tx_start_delay;			/*'pkt_data.count_us' time after which transmission will trigger */
        buff[3] = 0xFF & (count_trig >> 24);
        buff[4] = 0xFF & (count_trig >> 16);
        buff[5] = 0xFF & (count_trig >> 8);
        buff[6] = 0xFF &  count_trig;
    }
```
	
**doubt 2:**	

> 1. What is the use of lgw_fpga_get_tx_notch_delay() ? How we need to handle this for our custom board ?
> 2. From where "bw_delay_us = 1.5" value is fixed ?
> 3.  What is the significance of  "#define TX_START_DELAY_DEFAULT 1497 /* Calibrated value for 500KHz BW and notch filter disabled */" ? as LoRa gateway is for 125kHz then why this delay is added  because it is for 500KHz BW? How this value "1497" is derived ?

FILE NAME : "loragw_hal.c"

FNCTION NAME : "uint16_t lgw_get_tx_start_delay(bool tx_notch_enable, uint8_t bw))"

```

uint16_t lgw_get_tx_start_delay(bool tx_notch_enable, uint8_t bw) {
    float notch_delay_us = 0.0;
    float bw_delay_us = 0.0;
    float tx_start_delay;

    /* Notch filtering performed by FPGA adds a constant delay (group delay) that we need to compensate */
    if (tx_notch_enable) {
        notch_delay_us = lgw_fpga_get_tx_notch_delay();
    }

    /* Calibrated delay brought by SX1301 depending on signal bandwidth */
    switch (bw) {
        case BW_125KHZ:
            bw_delay_us = 1.5;
            break;
        case BW_500KHZ:
            /* Intended fall-through: it is the calibrated reference */
        default:
            break;
    }

    tx_start_delay = (float)TX_START_DELAY_DEFAULT - bw_delay_us - notch_delay_us;

    printf("INFO: tx_start_delay=%u (%f) - (%u, bw_delay=%f, notch_delay=%f)\n", (uint16_t)tx_start_delay, tx_start_delay, TX_START_DELAY_DEFAULT, bw_delay_us, notch_delay_us);

    return (uint16_t)tx_start_delay; /* keep truncating instead of rounding: better behaviour measured */
}
```

Thanks,
Gaurav

mcoracin, 2017-11-22:
Hello Gaurav,

If your implementing a CLass A gateway, you don't need to bother about those TX delay fine tuning, it is only useful for class B to have a very precise beacon transmission time.

This tx_delay is the time necessary for the radio to prepare itself before transmitting.
So when the HAL programs a TX based on the SX1301 timestamp, it needs to program a bit in advance to take into account the radio startup delay.

The tx_notch delay is only appliable for the reference design which has a FPGA, which is not the case for the iC880A concentrator, right?

In your case, I think that you can use a fixed delay of 1500µs always returned by the lgw_get_tx_start_delay() function.

Best regards

mcoracin, 2017-12-18:
as no answer, I close the issue

closed 2017-12-18 by mcoracin