Lora-net/lora_gateway #93: Restriction of values

Dieter-D, 2017-12-15:
Hi I have got a question to the allowed parameter settings for the mix_gain parameter

https://github.com/Lora-net/lora_gateway/blob/a955619271b5d0a46d32e08150acfbc1eed183b7/libloragw/src/loragw_hal.c#L674

Why does the code restrict values that are < 8?

In previous versions of the HAL (e.g. 3.1) there was a lower limit of 0.

This additional check is not being checked in the tx_utils so there is a wrong mixture of values that are displayed and values that are used within the HAL.

Is there any information about the gain settings available?

Thanks

mcoracin, 2017-12-18:
Hi,

The reason why the mix_gain value is restricted in the HAL is because below 8, the radio calibration is not functional, so TX performances are not optimal.

But you're right, the fact that util_tx_continuous does not report an error when using a mix_gain < 8 is an issue and should be fixed, thanks.

For gain settings available, you can refer to the global_conf.json file of the packet forwarder.

Best regards

Dieter-D, 2017-12-18:
Thanks for the fast response.
The problem is, that the util_tx_continuous utility does not check the result code of the "lgw_txgain_setconf()" function. So the HAL uses different values than the ones that are shown to the user.

Thanks.

closed 2017-12-18 by Dieter-D

Dieter-D, 2017-12-18:
Sorry I have got one follow up question:
Can the dig_gain parameter be varied between 0 and 3 without impact on the radio calibration?
The target is to compensate mix_gain settings < 8 accordingly.
Thanks.

mcoracin, 2017-12-18:
@Dieter-D , yes, I'll add the result code check in util_tx_continuous.

dig_gain can be varied between 0 and 3, yes.

Dieter-D, 2017-12-18:
ok!
Thanks gain!

ElizabethHoffman, 2017-12-18:
That is  an interesting point.
@mcoracin does the mix gain setting have any influence on the radio calibration?
Best regards.

mcoracin, 2017-12-19:
@ElizabethHoffman , not when you set the mix_gain within the allowed range.
The radio calibration is performed with a cross-talk mechanism of the SX1257s. If the mix_gain is too low (below 8), the signal is too weak to have enough cross-talk and a valid calibration.