Lora-net/lora_gateway #94: ERROR: Version of calibration firmware not expected, actual:66 expected:2

FrancoisP77, 2017-12-18:
Hello. 

I've got an issue with the loading of the firmware that I'd like to raise here. 

Here is my setup. 

- Lora Gateway version : 5.0.1
- Pi Zero 
- SX1301 board from the following Microchip kit (P/N: dv164140-1)
- Configuration file used : global_conf.json.PCB_E286.EU868.basic. 

I've crossed compiled the code successfuly and run it on the Pi Zero. 

I've run util_spi_stress with the following parameters -t 1. It does seem to work properly as the tool doesn't return any error. 

However, when I run lora_pkt_fwd I get the following error. 
**ERROR: Version of calibration firmware not expected, actual:66 expected:2**

Note that the version of the firmware is consistent (always 66). 

I tried both a dedicated power supply (5V 2A) and my computer to power up the board and the Pi but it didn't make any difference. 

I've been stuck on this issue for days now and I've really run out of ideas. 

Thank you for help.

mcoracin, 2017-12-18:
Hello, could you run the util_spi_stress test with -t 4 ?

It will test burst SPI read/write, which are used to load SX1301 firmwares.

FrancoisP77, 2017-12-18:
I ran the test and let it run. I got a mismatch after a few minutes. 
Could it be the wiring or some dark Microchip stuff ? I've checked the schematics but couldn't find anything odd.

mcoracin, 2017-12-18:
Yes, it could be the wiring.
Please refer to this post: https://github.com/Lora-net/packet_forwarder/issues/73#issuecomment-352389593

FrancoisP77, 2018-02-22:
I've managed to fix the issue by improving the wiring. 
Thanks for your help.

closed 2018-02-22 by FrancoisP77

cdupont, 2020-11-09:
I had this issue, I found that lowering the SPI_SPEED to 1000000 or 2000000 helped. https://github.com/Lora-net/lora_gateway/blob/master/libloragw/src/loragw_spi.native.c#L56