Lora-net/lora_gateway #95: ERROR: TX gain LUT: SX1757 mixer gains < 8 are not supported

FrancoisP77, 2017-12-18:
Hello. 

I'd like to raise an issue with the following configuration file global_conf.json.PCB_E286.EU868.basic. 

When I use this configuration I get the following error. 
**ERROR: TX gain LUT: SX1757 mixer gains < 8 are not supported**

I've double checked the values in each mix_gain field but they all seem to be coherent and > 8. 
The Lora gateway loads it though when I remove all the tx_lut entries. 

Is this expected with default configuration files ?

Regards.

mcoracin, 2017-12-18:
Hello,

I don't understand, do you get this error when using the global_conf.json.PCB_E286.EU868.basic as released, or have you modified it?
Indeed, mix_gain as to be in the range of [8..15]. If it is not the case, an error is raised. But the global_conf.json.PCB_E286.EU868.basic released does not contain any tx_lut entry with mix_gain < 8, as far as I can see.

If you remove the tx_lut entries, the packet forwarder can start using default tx_lut parameters, but you may have bad TX performances.

The error message you put says "SX1757 mixer gains...", is it a typo? It should be SX1257.

FrancoisP77, 2017-12-18:
Thank you for the quick response. 

This is a typo. I'll edit my post. 

I did use the global_conf.json.PCB_E286.EU868.basic file as released then removed the following entries to see if the gateway loaded the configuration so it would allow me to locate the issue. 

```
...
        "tx_lut_0": {
            /* TX gain table, index 0 */
            "pa_gain": 0,
            "mix_gain": 8,
            "rf_power": -6,
            "dig_gain": 0
},
...
```

mcoracin, 2017-12-18:
Do you mean that using the latest lora_gateway and packet_forwarder version with the global_conf.json.PCB_E286.EU868.basic "as released" you get an error when launching the packet forwarder?

It should not. :) I've just tested it to cross check.

A mix_gain of 8 is allowed.

Please elaborate on your configuration.

FrancoisP77, 2017-12-18:
Yes that's it.
I cloned the repository from GiHub last week and did a copy of the file.

`cp ./cfg/global_conf.json.PCB_E286.EU868.basic ./global_conf.json`

FrancoisP77, 2017-12-18:
I ran the gateway on my dev environment and it loads the configuration file. I may have a problem with my cross compilation toolchain. (Glibc versions etc.).

I will double check and come back to you.

FrancoisP77, 2018-02-22:
I had an issue with my cross compilation toolchain. Both the kernel and glibc versions were wrong. 
Issue's fixed now.

closed 2018-02-22 by FrancoisP77