Lora-net/lora_gateway #96: Failed to load fw 1

zhoubx123, 2017-12-26:
HI , who can help me ?
ERROR: Failed to load fw 1
ERROR: Version of calibration firmware not expected, actual:49 expected:2
ERROR: [main] failed to start the concentrator

Theoi-Meteoroi, 2017-12-26:
The first line is pretty clear. Firmware failed to load.
The second line complains about calibration firmware levels. You have the wrong firmware set for install.
The third line becomes obvious. Party over. Not today.

3nd0y, 2018-01-08:
@Theoi-Meteoroi which firmware did you mean? Is the concentrator has a firmware inside? I have similar issue but with error: actual:0 expected:2

Dieter-D, 2018-01-08:
@zhoubx123 
Did you execute the reset script (https://github.com/Lora-net/lora_gateway/blob/master/reset_lgw.sh)  before starting the software?

zhoubx123, 2018-01-10:
Thank you for  reply.
 My problem has been solved. when I shorten the SPI line and connect more ground lines.

closed 2018-01-10 by zhoubx123