Lora-net/lora_gateway #97: SX1301 Reg Test

rego21, 2018-01-23:
Hello all,

Im trying to connect the SX1301 to on STM32L073 Nucleo. I ran the reg test and obtained the following results (image) :
[image]
Should i worry about this or anyone knows how to solve the problem? Because only have problems on two of them...

Thank you all!

FlorianMarcon, 2019-10-10:
Hi @rego21 !

Have you find a solution ?

rego21, 2019-10-19:
> 
> 
> Hi @rego21 !
> 
> Have you find a solution ?

Hi @FlorianMarcon 

Sorry for late response, i tried using STM32L073 but its not possible due to the lack of memory. You can try with a larger (memory) STM, it should work with any problems. ARM embedded has everything you need! 

Best regards!

closed 2021-07-08 by rbaldwin13