Lora-net/lora_gateway #102: Multiple SPI ports, howto?

rgillan, 2018-02-20:
We are trying to run two SX1301s off a common SPI bus (spidev0.0, spidev0.1).  In loragw_spi.native.c it states that it could be used with multiple SPI ports in parallel (explicit file descriptor).

Can anyone give some guidance as to how we could do this?

mcoracin, 2018-02-20:
This has never been tested with multiple SPI ports, and would require some changes because currently the SPI port to be used is hardcoded in the lgw_spi_open() function to use SPI_DEV_PATH.
But I think, the lgw_spi_open() function could be updated to take a PATH in parameter and use it to open the specified port. Then the top application would access one port or the other with different spi_target pointers?

rgillan, 2018-02-20:
So continuing, the specific SPI_DEV_PATH would need to be passed from lgw_connect in loragw_reg, and then to there from lgw_start in loragw_hal, and then from the upper level call in the packet forwarder.  That way there would only need to be a single libloragw.   I might experiment and see how it goes (and report back here).  Thanks for the prompt response, it's appreciated.

dovetion, 2018-04-10:
Hey, I've done that, now I can use two SX1301 in a gateway board.
You just need add a parameter in **almost** every function in HAL, and call `lgw_start(int spi_port)` twice.
But the difficult part is how to design a good protocol to use that, and I am still working on it.
If you have same problem or good advice I'd like to have some chat :)

mcoracin, 2018-11-16:
It seems that the initial issue has been answered, so I close this issue. Thank you.

closed 2018-11-16 by mcoracin

JayKickliter, 2019-05-22:
I know this issue is long closed, but I recently started using this code base. 

In response to:
> You just need add a parameter in almost every function in HAL, and call lgw_start(int spi_port) twice.

This library _should_ have done this from the beginning but in the form of a `struct lgw` object. `lgw` would also hold user provided SPI read/write (decoupling the lib from linux spidev), and logging functions. But now I'm stuck with a mess of global state in a codebase I have to use because the chip is undocumented.