Lora-net/lora_gateway #122: SX1301 How to set register Low Data Rate Optimization

fanjm95, 2018-09-07:
Hi,
    I am testing the sensitivity of the gateway. SF12 sent by the machine always shows CRC validation error.I think the gateway should also be turned on for low datarate optimization.How to turned on  low datarate optimization?

fanjm95, 2018-09-07:
.How to turned off low datarate optimization?

fanjm95, 2018-09-07:
[image]

fanjm95, 2018-09-07:
I want to disenable low data rate optimization.but I do not know where to disenable?

Dollar-Yao, 2018-09-19:
> I want to disenable low data rate optimization.but I do not know where to disenable?

Hi, you can search 'SET_PPM_ON' and find the definition,  the value of 'lora_rx_ppm_offset' is control the low data rate optimization.

mcoracin, 2018-11-16:
It is enabled by default in the HAL for SF11/SF12:
https://github.com/Lora-net/lora_gateway/blob/a955619271b5d0a46d32e08150acfbc1eed183b7/libloragw/src/loragw_hal.c#L929

closed 2018-11-16 by mcoracin