Lora-net/lora_gateway #127: Get "CRC error, damaged packet"

wx-ta, 2018-11-14:
Get "CRC error, damaged packet",when run ./test_loragw_rx 435.425 410.425,only 2% packets are received correctly,who known why?

mcoracin, 2018-11-14:
Using private/public syncword on both sides? gateway and end-node?

wx-ta, 2018-11-15:
thinks  for your reply, gateway side,library.cfg file,i config "CFG_NET= private"; end_node side:i am using a sx127x send packets,,both them using bw125k.  how to set syncword in lora mode?

mcoracin, 2018-11-15:
There is no CFG_NET option in the standard library.cfg, so I'm not sure which version you are using. But in the libraloragw library, you have to set 'lorawan_public' of the board configuration to true or false depending on the configuration you are using on the end-note.
But I don't kniw which lorawan stack or sx127x driver you are using either...

wx-ta, 2018-11-15:
Sorry, i use an old version of libloragw clone from https://github.com/RAKWireless/RAK831_LoRaGateway , because i use usb cable(ft2232h usb to spi chip) connect sx1301 board to pc.
i saw the new version has remove usb2spi bridge part. so,i copy the old ftdi support file xspi.ftdt.c to new version src/ folder ,then modify it and the Makefile, i get the new version program whit usb2spi support, you told me to config lorawan_public = false,i did it,but it cannot receive any packet. new program can send packet correctly.
this is the log:
[root@localhost libloragw]# ./test_loragw_hal -a 435.625 -b 410.625 -t 435.625 -r 1255
Beginning of test for loragw_hal.c
*** Library version information ***
Version: 5.0.1;

lgw_board_setconf:427: Note: board configuration; lorawan_public:0, clksrc:1
Note: board configuration; lorawan_public:0, clksrc:1
lgw_rxrf_setconf:489: Note: rf_chain 0 configuration; en:1 freq:435625000 rssi_offset:0.000000 radio_type:1 tx_enable:1 tx_notch_freq:129000
lgw_rxrf_setconf:489: Note: rf_chain 1 configuration; en:1 freq:410625000 rssi_offset:0.000000 radio_type:1 tx_enable:0 tx_notch_freq:129000
lgw_rxif_setconf:608: Note: LoRa 'multi' if_chain 0 configuration; en:1 freq:-400000 SF_mask:0x7e
lgw_rxif_setconf:608: Note: LoRa 'multi' if_chain 1 configuration; en:1 freq:-200000 SF_mask:0x7e
lgw_rxif_setconf:608: Note: LoRa 'multi' if_chain 2 configuration; en:1 freq:0 SF_mask:0x7e
lgw_rxif_setconf:608: Note: LoRa 'multi' if_chain 3 configuration; en:1 freq:-400000 SF_mask:0x7e
lgw_rxif_setconf:608: Note: LoRa 'multi' if_chain 4 configuration; en:1 freq:-200000 SF_mask:0x7e
lgw_rxif_setconf:608: Note: LoRa 'multi' if_chain 5 configuration; en:1 freq:0 SF_mask:0x7e
lgw_rxif_setconf:608: Note: LoRa 'multi' if_chain 6 configuration; en:1 freq:200000 SF_mask:0x7e
lgw_rxif_setconf:608: Note: LoRa 'multi' if_chain 7 configuration; en:1 freq:400000 SF_mask:0x7e
lgw_rxif_setconf:582: Note: LoRa 'std' if_chain 8 configuration; en:1 freq:0 bw:2 dr:16
lgw_rxif_setconf:638: Note: FSK if_chain 9 configuration; en:1 freq:0 bw:2 dr:64000 (64000 real dr) sync:0xC194C1
lgw_start:824: Note: calibration started (time: 2300 ms)
lgw_start:845: Note: calibration finished (status = 175)
WARNING: problem in calibration of radio B for image rejection
Info: Initialising AGC firmware...
Info: putting back original RADIO_SELECT value
*** Concentrator started ***
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
INFO: Enabling TX notch filter
59.d0.0.0.0.0.0.0.0.99.14.0.0.6.0.0.54.58.2e.54.45.53.54.2e.4c.4f.52.41.2e.47.57.2e.0.0.0.0.end
+++
Sending packet #0, rf path 0, return 0
status -> 4:2:
TX finished
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
INFO: Enabling TX notch filter
59.d0.0.0.0.0.0.0.0.99.14.0.0.6.0.0.54.58.2e.54.45.53.54.2e.4c.4f.52.41.2e.47.57.2e.0.0.0.1.end
+++
Sending packet #1, rf path 0, return 0
status -> 4:2:
TX finished

wx-ta, 2018-11-16:
I use sx127x driver  clone from https://github.com/fifteenhex/sx127x ， there is no syncword config code，i found that sx127x syncword register address is 0x39 in lora mode from https://github.com/sinny777/hukam/blob/master/microcontrollers/ESP32/HukamESP32LoraBoard/libraries/LoRa/src/LoRa.cpp  line：28
I find syncword config register on sx1301，and config “lorawan_public”  to false.
now，I can received packet sx3101 send out itself，cannot receive packer send from sx127x yet.

wx-ta, 2018-11-16:
I am sure that sx127x syncword register default value is 0x12, private network. because i read this register after my program run.
root@myimx6ek140p:~# cat /sys/class/sx1278/sx1278-0/syncword 
18

(DEC)18   =   (HEX)0x12

wx-ta, 2018-11-16:
Thank mcoracin, I find these reasons with your help;
1. "lorawan_public "should be set false,then sx1301 will be set syncword with 0x12;
2 . CRC error,the reason is sx127x endnode CRC is not enabled;
Thank you !

closed 2018-11-16 by wx-ta