Lora-net/lora_gateway #130: Some problems occured in AM335x with SX1301

suyiii, 2019-01-20:
Hello!

I use AM335x as MCU which is connected with SX1301 via SPI, RESET.
Then  compile the project in Ubuntu and copy files including reset_lgw.sh, lora_pkt_fwd, *conf_json and update_gwid.sh to AM335x.

I update the reset pin and change sleep 0.1 to sleep 1 in reset_lgw.sh because sleep < float > is not supported in my platform.

After reset the SX1301, I excute ./lora_pkt_fwd and error info occur in the log:

lgw_spi_wb:310: BURST WRITE: to trans 8192 # chunk 1024 # transferred 1024 
lgw_spi_wb:310: BURST WRITE: to trans 7168 # chunk 1024 # transferred 2048 
lgw_spi_wb:310: BURST WRITE: to trans 6144 # chunk 1024 # transferred 3072 
lgw_spi_wb:310: BURST WRITE: to trans 5120 # chunk 1024 # transferred 4096 
lgw_spi_wb:310: BURST WRITE: to trans 4096 # chunk 1024 # transferred 5120 
lgw_spi_wb:310: BURST WRITE: to trans 3072 # chunk 1024 # transferred 6144 
lgw_spi_wb:310: BURST WRITE: to trans 2048 # chunk 1024 # transferred 7168 
lgw_spi_wb:310: BURST WRITE: to trans 1024 # chunk 1024 # transferred 8192 
Note: SPI burst write success
Note: SPI read success
lgw_spi_rb:372: BURST READ: to trans 8192 # chunk 1024 # transferred 1024 
lgw_spi_rb:372: BURST READ: to trans 7168 # chunk 1024 # transferred 2048 
lgw_spi_rb:372: BURST READ: to trans 6144 # chunk 1024 # transferred 3072 
lgw_spi_rb:372: BURST READ: to trans 5120 # chunk 1024 # transferred 4096 
lgw_spi_rb:372: BURST READ: to trans 4096 # chunk 1024 # transferred 5120 
lgw_spi_rb:372: BURST READ: to trans 3072 # chunk 1024 # transferred 6144 
lgw_spi_rb:372: BURST READ: to trans 2048 # chunk 1024 # transferred 7168 
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 8192 
Note: SPI burst read success
ERROR: Failed to load fw 1
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
ERROR: Version of calibration firmware not expected, actual:15 expected:2
ERROR: [main] failed to start the concentrator

I searched the log in github. Mostly told me to reset the pin and check the hardware. So I confirmed the reset pin has been reseted and spi has been connected correctly because I have tested it by test_loragw_spi and test_loragw_reg and 48 in test_loragw_spi and  mostly MATCH in test_loragw_reg.

Additionally,  I copied the excute file to PI3, of course with spi path adpted, and successed, no matter which vesion of compiler. The AM335x uses arm-linux-gnueabihf-gcc_4.7 to compile the kernel, and PI3 has arm-linux-gnueabihf-gcc_4.9. So I compiled the project with both version. As a result, PI3 can run both version correctly, but AM335x neither.

So, how can I run the program correctly in AM335x platform?

suyiii, 2019-01-20:
I excute ./util_spi_stress -t 4 and all success.

Log as follow:

lgw_spi_wb:310: BURST WRITE: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst read success
Cycle 2210 > did a 1024-byte R/W on a data buffer with no error
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst read success
Cycle 2211 > did a 1024-byte R/W on a data buffer with no error
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst read success
Cycle 2212 > did a 1024-byte R/W on a data buffer with no error
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst read success
Cycle 2213 > did a 1024-byte R/W on a data buffer with no error
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst read success
Cycle 2214 > did a 1024-byte R/W on a data buffer with no error
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst read success
Cycle 2215 > did a 1024-byte R/W on a data buffer with no error
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst write success
lgw_spi_wb:310: BURST WRITE: to trans 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst rs 2 # chunk 2 # transferred 2 
Note: SPI burst write success
lgw_spi_rb:372: BURST READ: to trans 1024 # chunk 1024 # transferred 1024 
Note: SPI burst read success
Cycle 2223 > did a 1024-byte R/W on a data buffer with no error
Note: SPI port closed
Note: success disconnecting the concentrator
INFO: Exiting LoRa concentrator SPI stress-test program

yehming, 2019-09-16:
Hello 
I am using the am335x platform as the LoRaWAN gateway and it is currently working well.
The lora_pkt_fwd program is working well.

What is the output log when you run the following 3 programs, Is it convenient to provide logs?
1.    lora_gateway/libloragw/test_loragw_spi
2.    lora_gateway/util_spi_stress/util_spi_stress
3.    lora_gateway/libloragw/test_loragw_reg

I can see if I can help you.

closed 2021-07-08 by rbaldwin13