Lora-net/lora_gateway #137: Can this code receive non Lorawan packets?

EoinBullock, 2019-04-06:
I have a feeling this is why I cannot receive fsk or lora packets. My payload is completely custom with no Lorawan stuff. Is this why no packets are being logged?

If this is the case how can I modify the Sx1301 firmware to log every packet and not just Lorawan?

reissjason, 2019-04-08:
There are CRC and sync word options to filter packets. This is the only filter encoded in a gateway.
Only packets matching the sync_word will be reported.

LoRa sync_word 1-byte 0x34 Public, 0x12 Private
FSK sync_word 3-bytes 0xC194C1

EoinBullock, 2019-04-08:
I have tried all the different variations and no matter what I do, my packets are not logged! As soon as I switch to LoraWan payload format it works and my packets are logged. I have a feeling that the firmware is looking for specific LoraWan bytes in the payload.

Normally when a LoraWan end node sends a 4 bytes payload, there is a whole load of extra Lorawan stuff added into the payload in the mac header.

If you are sure it is not filtering my packets, if I write 0,1,2,3 into the TX payload FIFO and send it, do you think the gateway will log it? Assume I have the correct sync word etc

reissjason, 2019-04-08:
There is also an I/Q polarity setting based on uplink/downlink packet direction so gateways do not hear other gateways and likewise for devices.

Downlinks are sent with "ipol":true in the JSON to the packet forwarder to invert the I/Q setting to match the end-device Rx window properties.

The gateway and forwarder do not verify the payload content.
They have no notion of a LoRaWAN header or MIC added within.

EoinBullock, 2019-04-08:
Ok thanks for the clarification.
So if I want to be able to send packets to the gateway, what are the settings I should use?
Implicit or explicit?
Which SF?
Bandwidth 125kHz I assume?
Payload Crc or not?
Which coding rate?

reissjason, 2019-04-08:
Same as a LoRaWAN end-device, see Radio.SetTxConfig
https://github.com/Lora-net/LoRaMac-node/blob/develop/src/mac/region/RegionUS915.c

EoinBullock, 2019-04-08:
I tried to figure it out from the code but I got lost, my gateway is in 868 band.
I have tried the following settings on the SX1276 but no luck
`	RadioSettings.LoraSettings.CrcOn=1;
	RadioSettings.LoraSettings.SpreadFactor=7;
	RadioSettings.LoraSettings.Coderate=1;
	RadioSettings.LoraSettings.PreambleLen=6;
	RadioSettings.LoraSettings.Bandwidth=BW_125_KHZ;
	RadioSettings.LoraSettings.Implicit=0;
	RadioSettings.LoraSettings.PayloadLen=16;`

frequency 868.3MHz

The .json file of my gateway looks like this:

`{
    "SX1301_conf": {
        "lorawan_public": true,
        "clksrc": 1, /* radio_1 provides clock to concentrator */
        "radio_0": {
            "enable": true,
            "type": "SX1257",
            "freq": 867500000,
            "rssi_offset": -166.0,
            "tx_enable": false
        },
        "radio_1": {
            "enable": true,
            "type": "SX1257",
            "freq": 868500000,
            "rssi_offset": -166.0,
            "tx_enable": false
        },
        "chan_multiSF_0": {
            /* Lora MAC channel, 125kHz, all SF, 868.1 MHz */
            "enable": true,
            "radio": 1,
            "if": -400000
        },
        "chan_multiSF_1": {
            /* Lora MAC channel, 125kHz, all SF, 868.3 MHz */
            "enable": true,
            "radio": 1,
            "if": -200000
        },
        "chan_multiSF_2": {
            /* Lora MAC channel, 125kHz, all SF, 868.5 MHz */
            "enable": true,
            "radio": 1,
            "if": 0
        },
        "chan_multiSF_3": {
            /* Lora MAC channel, 125kHz, all SF, 867.1 MHz */
            "enable": true,
            "radio": 0,
            "if": -400000
        },
        "chan_multiSF_4": {
            /* Lora MAC channel, 125kHz, all SF, 867.3 MHz */
            "enable": true,
            "radio": 0,
            "if": -200000
        },
        "chan_multiSF_5": {
            /* Lora MAC channel, 125kHz, all SF, 867.5 MHz */
            "enable": true,
            "radio": 0,
            "if": 0
        },
        "chan_multiSF_6": {
            /* Lora MAC channel, 125kHz, all SF, 867.7 MHz */
            "enable": true,
            "radio": 0,
            "if": 200000
        },
        "chan_multiSF_7": {
            /* Lora MAC channel, 125kHz, all SF, 867.9 MHz */
            "enable": true,
            "radio": 0,
            "if": 400000
        },
        "chan_Lora_std": {
            /* Lora MAC channel, 250kHz, SF7, 868.3 MHz */
            "enable": true,
            "radio": 1,
            "if": -200000,
            "bandwidth": 250000,
            "spread_factor": 7
        },
        "chan_FSK": {
            /* FSK 50kbps channel, 868.8 MHz */
            "enable": true,
            "radio": 1,
            "if": 300000,
            "bandwidth": 125000,
            "datarate": 50000
        }
    },
    "gateway_conf": {
        "gateway_ID": "AA555A0000000000"
    }
}
`

EoinBullock, 2019-04-09:
I have had some luck with this!

When I transmit on SF7-SF10 the gateway logs my packets, well actually it logs 2 entries 300ms apart for each packet, but I will debug that later.

When I try on SF11 and SF12 the gateway constantly shows CRC_BAD, in SF11 and SF12 I have the low data rate optimise bit set on the end node.

Any ideas why I would get CRC_BAD for SF11 and SF12?
Eoin

Dani99l, 2019-04-11:
@EoinBullock  can you specify which murata module are you using and which lora libs ?

EoinBullock, 2019-04-11:
The dev kit I am using is the B-LO72Z-LRWAN1 (https://www.st.com/en/evaluation-tools/b-l072z-lrwan1.html)
The code is my own modified version of the end-node code found in this repository LoraMacNode.
It's just odd that everything is fine except in SF11 and SF12, my settings are:

BW 125kHz
Code rate 1
Explicit
CRC On
I have tried all the channels too.

But I still get CRC BAD

closed 2021-07-08 by rbaldwin13